rfquack/RFQuack #2: RFQuak ESP32 w/CC1101

dgeppert, 2020-10-12:
Using the instructions in Git I managed to flash the ESP32 using the rfquack/rfquack docker and connect via rfquack/cli docker:

```
docker run --device /dev/ttyUSB0:/board -e BOARD=ESP32 -e RADIOA=CC1101 -e RADIOA_CS=5 -e RADIOA_IRQ=4 --rm -it rfquack/rfquack
docker run --device /dev/ttyUSB0 --user root --rm -it rfquack/cli -l DEBUG tty -P /dev/ttyUSB0

```
However, the CC1101 connected via the ESP32 VSPI seems not to be correctly initialised, the cli only reports _Select a dongle typing: q.dongle(id)_ i.e. no radioA object is available.

Using other software I manage to communicate with the CC1101 and exactly the same setup, so I wonder if RFQuack uses a different way to connect the ESP SPI? From the instructions and the photos I found, I unfortunately could not deduct the cabling.

```
[dirk@t520 RFQuack]$ docker -D run --device=/dev/ttyUSB0 --user=root --rm -it rfquack/cli -l DEBUG tty -P /dev/ttyUSB0 -b 115200 -s 8 -p N -S 1

(Banner message here)

2020-10-12 06:58:26 ab141e45367c rfquack.transport[1] INFO Port opened

  Select a dongle typing: q.dongle(id)
2020-10-12 06:58:26 ab141e45367c rfquack.transport[1] DEBUG b'any/in/set/ping/rfquack_VoidValue/ping' (0 bytes)
2020-10-12 06:58:26 ab141e45367c rfquack.transport[1] DEBUG Writing packet = b'>any/in/set/ping/rfquack_VoidValue/ping~\x00'

(here disconnecting the USB)

2020-10-12 06:58:38 ab141e45367c rfquack.transport[1] ERROR read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
RFQuack(/dev/ttyUSB0)>                                                                                                    

(and connecting it back)

RFQuack(/dev/ttyUSB0)>      

```
q.[tab completion] only proposes: 

[image]

connecting via screen and pressing RESET show:

```
ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[RFQK]        137 T: Standalone mode network setup: done!
[R�[RFQK]        463 T: Serial transport connected
[RFQK]        564 T: Serial transport connected
[RFQK]      10665 E: Unable to initialize radio, got code -2
[RFQK]      10665 T: Something went wrong, check your wiring.
```

I wired up ESP32 VSPI (MOSI GPIO 23, MISO GPIO 19, SCLK GPIO 18, CS GPIO 5, GDO0 GPIO 4).
which SPI bus RFQuak is using? seems there are three; SPI (used for flash, so usually not recommended), VSPI and HSPI

as far as I understand all three SPI, VSPI, HSPI are h/w SPI.

currently I am using VSPI, last night I tried SPI but no success, maybe try again. **which SPI bus i.e. GPIOs RFQuak using?**

Devkit-v4
```
        MOSI MISO  WP   HD  CLK   CS
         ID    Q   WP   HD  CLK  CS0
VSPI     23   19   22   21   18    5
HSPI     13   12    2    4   14   15
 SPI      8    7   10    9    6   11

```
Using a second set of h/w results in the same problem.

Thanks in advance for your help on this issue!

Regads, Dirk

phretor, 2020-10-12:
Hi @dgeppert - I'd suggest to try the HSPI.

@Guglio95 you used RFQuack with an ESP32: do you by any chance recall or have a picture of your SPI wiring?

dgeppert, 2020-10-12:
same with SPI and HSPI:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[RFQK]        137 T: Standalone mode network setup: done!
[R�[RFQK]        463 T: Serial transport connected
[RFQK]        564 T: Serial transport connected
[RFQK]      10665 E: Unable to initialize radio, got code -2
[RFQK]      10665 T: Something went wrong, check your wiring.
```

phretor, 2020-10-12:
@dgeppert can you edit the `docker/my-endpoint.sh` as follows:

```
#!/bin/sh
set -e

export UUID=$(python  -c 'import uuid; print(uuid.uuid4())')
echo "Installation ID is $UUID"

# Compile template
j2 /quack/src/main.cpp.j2 > /quack/src/main.cpp

if [ -z "$BOARD" ]
then
	echo "No board selected, defaults to HUZZAH32"
	BOARD="HUZZAH32"
fi

cat /quack/src/main.cpp
```

and run the `docker run ...` again? Nothing will happen and you'll see a lot of output: that way we can see what `.cpp` code is being compiled and uploaded on the dongle.

dgeppert, 2020-10-12:
replaced the last line in `docker/my-endpoint.sh`

```
# platformio run -t upload -e $BOARD
cat /quack/src/main.cpp
```
and ran 

```
docker run --device /dev/ttyUSB0:/board -e BOARD=ESP32 -e RADIOA=CC1101 -e RADIOA_CS=5 -e RADIOA_IRQ=4 --rm -it rfquack/rfquack
```

but strangely do not see the effect of `cat main.cpp`, neither using `screen` nor `docker run ... rfquack/cli`

maybe I misunderstood?

dgeppert, 2020-10-12:
btw I only downloaded your docker, did not build it

phretor, 2020-10-12:
Do you have any other way to see the content of main.cpp?

dgeppert, 2020-10-12:
sorry, I do not know where to find it.

btw I  build RFQuak and RFQuack-cli from source, I up to now have problems to get e.g.
RFQuack/examples/RFQuack-esp32-CC1101-serial/ to work (well I am new to platformio)

and I noticed that in this example the platformio.ini points to the old v1 whereas I use a v4:
board = esp32doit-devkit-v1

maybe this was used to build the docker container and this causes the problem?

Guglio95, 2020-10-12:
Ciao!
The docker image assumes your ESP32 board is a [featheresp32](https://www.adafruit.com/product/3405). Are you using a different model?
Platformio, based on the board definition, includes a header file which determines the pinout to be used for the SPI bus.

dgeppert, 2020-10-12:
ciao Guglio, yes I am using a newer devkit v4 : az-delivery-devkit-v4

I am trying to build from an example using platformio, but need some time to learn
currently I get compiling errors on:

```
RFQuack/src/radio/RFQCC1101.h
RadioLib/src/modules/RF69/RF69.cpp
RadioLib/src/modules/RF69/RF69.cpp
```

I do not want to get this issue too long, maybe I will come back later

unless you do have some other idea
maybe this will solve the problem

phretor, 2020-10-12:
@dgeppert using PlatformIO directly is best because running PlatformIO into Docker will give you less control. Can you share the error messages you mentioned? No problem for making this issue long ;-) the longer it gets, the more informative it will be for the future!

dgeppert, 2020-10-12:
Hello @phretor

well you asked for it and it happened that I just prepared this post ;-) here we go:

```
# ====================================================================================================================================

[dirk@t520 platformio]$ rm -rf ~/.platformio/

[dirk@t520 platformio]$ python get-platformio.py
Installer version: 0.3.5
Platform: Linux-5.8.7-200.fc32.x86_64-x86_64-with-glibc2.2.5
Python version: 3.8.5 (default, Aug 12 2020, 00:00:00) 
[GCC 10.2.1 20200723 (Red Hat 10.2.1-1)]
Python path: /usr/bin/python
Creating a virtual environment at /home/dirk/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Processing /home/dirk/.cache/pip/wheels/73/62/c9/001bd2b98220441af507fa7a05f8bb44b142bfdb9e7d8f3e99/platformio-5.0.1-py3-none-any.whl
Collecting marshmallow>=2
  Using cached marshmallow-3.8.0-py2.py3-none-any.whl (46 kB)
Collecting tabulate<1,>=0.8.3
  Using cached tabulate-0.8.7-py3-none-any.whl (24 kB)
Collecting pyelftools<1,>=0.25
  Using cached pyelftools-0.26-py2.py3-none-any.whl (136 kB)
Collecting bottle<0.13
  Using cached bottle-0.12.18-py3-none-any.whl (89 kB)
Collecting semantic-version<3,>=2.8.1
  Using cached semantic_version-2.8.5-py2.py3-none-any.whl (15 kB)
Collecting click<8,>=5
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Collecting requests<3,>=2.4.0
  Using cached requests-2.24.0-py2.py3-none-any.whl (61 kB)
Collecting colorama
  Using cached colorama-0.4.3-py2.py3-none-any.whl (15 kB)
Collecting pyserial!=3.3,<4,>=3
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting chardet<4,>=3.0.2
  Using cached chardet-3.0.4-py2.py3-none-any.whl (133 kB)
Collecting urllib3!=1.25.0,!=1.25.1,<1.26,>=1.21.1
  Using cached urllib3-1.25.10-py2.py3-none-any.whl (127 kB)
Collecting certifi>=2017.4.17
  Using cached certifi-2020.6.20-py2.py3-none-any.whl (156 kB)
Collecting idna<3,>=2.5
  Using cached idna-2.10-py2.py3-none-any.whl (58 kB)
Installing collected packages: marshmallow, tabulate, pyelftools, bottle, semantic-version, click, chardet, urllib3, certifi, idna, requests, colorama, pyserial, platformio
Successfully installed bottle-0.12.18 certifi-2020.6.20 chardet-3.0.4 click-7.1.2 colorama-0.4.3 idna-2.10 marshmallow-3.8.0 platformio-5.0.1 pyelftools-0.26 pyserial-3.4 requests-2.24.0 semantic-version-2.8.5 tabulate-0.8.7 urllib3-1.25.10

PlatformIO Core has been successfully installed into an isolated environment `/home/dirk/.platformio/penv`!

The full path to `platformio.exe` is `/home/dirk/.platformio/penv/bin/platformio`

If you need an access to `platformio.exe` from other applications, please install Shell Commands
(add PlatformIO Core binary directory `/home/dirk/.platformio/penv/bin` to the system environment PATH variable):

See https://docs.platformio.org/page/installation.html#install-shell-commands

[dirk@t520 platformio]$ sudo ln -sf ~/.platformio/penv/bin/platformio /usr/local/bin/platformio
[dirk@t520 platformio]$ sudo ln -sf ~/.platformio/penv/bin/pio /usr/local/bin/pio
[dirk@t520 platformio]$ sudo ln -sf ~/.platformio/penv/bin/piodebuggdb /usr/local/bin/piodebuggdb

# ====================================================================================================================================

[dirk@t520 platformio]$ platformio lib -g install git+https://github.com/nanopb/nanopb.git#0.3.9.2
Library Storage: /home/dirk/.platformio/lib
Library Manager: Installing git+https://github.com/nanopb/nanopb.git#0.3.9.2
git version 2.26.2
Cloning into '/home/dirk/.platformio/.cache/tmp/pkg-installing-f13rqg41'...
[... 155 lines cut ...]

[dirk@t520 RFQuack-esp32-CC1101-serial]$ pio boards ESP32
Platform: espressif32
==========================================================================================================================
ID                               MCU    Frequency    Flash    RAM     Name
-------------------------------  -----  -----------  -------  ------  --------------------------------
esp32cam                         ESP32  240MHz       4MB      320KB   AI Thinker ESP32-CAM
az-delivery-devkit-v4            ESP32  240MHz       16MB     520KB   AZ-Delivery ESP-32 Dev Kit C V4
featheresp32                     ESP32  240MHz       4MB      320KB   Adafruit ESP32 Feather
esp32doit-devkit-v1              ESP32  240MHz       4MB      320KB   DOIT ESP32 DEVKIT V1
...

# ====================================================================================================================================

[dirk@t520 RFQuack-esp32-CC1101-serial]$ pio run -e ESP32 > piorun001.log 2>&1 

see log file!

# ====================================================================================================================================
```
here the log file:

[piorun001.log](https://github.com/rfquack/RFQuack/files/5365885/piorun001.log)

thanks for your help!

phretor, 2020-10-12:
I forgot to mention that if you go the PlatformIO route there are a couple of oddities you should be careful of. There are multiple errors there, but let's try to rule them out one at a time:

 - the "NC" macro is not defined (see the log around `RadioA radioA = new Module(2, 21, NC);`) so I'm guessing it's part of the board's definition files: try to change that "esp32doit-devkit-v1" into "*-v4" (as you mentioned you have a V4 board. This likely won't help, but it's worth a try.

 - I can't guarantee that https://github.com/rfquack/rfquack and https://github.com/trendmicro/rfquack are aligned: please try the former, which is the main one (I'll put a redirect for the future: thanks for spotting that!)

 - make sure that *on your system* you install the Protobuf dev libraries, and that they're version-aligned with what Python installs: this is needed because the *.proto file will be compiled into *.h(pp) and *.py files that will later on be compiled into the executable objects that make the firmware

dgeppert, 2020-10-12:
gracie, I will give it a try

for waht concerns the last bullet, does this test make sense and the versions seem ok?

[dirk@t520 RFQuack]$ rpm -qa | grep protobuf
protobuf-lite-3.11.2-2.fc32.x86_64
protobuf-3.11.2-2.fc32.x86_64
protobuf-compiler-3.11.2-2.fc32.x86_64
python3-protobuf-3.11.2-2.fc32.noarch
protobuf-c-1.3.2-2.fc32.x86_64
protobuf-devel-3.11.2-2.fc32.x86_64

phretor, 2020-10-12:
Sort of OK. The last bit to check is that `make proto` from the main RFQuack directory (i.e., this cloned repository) does not return errors. Feel free to post the entire output here.

dgeppert, 2020-10-12:
scratched and reinstalled platformio as before, now checking out the official version:

``` 
[dirk@t520 esp]$ git clone https://github.com/rfquack/RFQuack.git
Cloning into 'RFQuack'...
remote: Enumerating objects: 1058, done.
remote: Counting objects: 100% (1058/1058), done.
remote: Compressing objects: 100% (489/489), done.
remote: Total 1058 (delta 585), reused 1002 (delta 530), pack-reused 0
Receiving objects: 100% (1058/1058), 24.56 MiB | 2.90 MiB/s, done.
Resolving deltas: 100% (585/585), done.

[dirk@t520 esp]$ cd RFQuack

[dirk@t520 RFQuack]$ make proto
cd "/home/dirk/.platformio/lib/Nanopb/generator/proto" ;  make
make[1]: Entering directory '/home/dirk/.platformio/lib/Nanopb/generator/proto'
protoc --python_out=. nanopb.proto
protoc --python_out=. plugin.proto
make[1]: Leaving directory '/home/dirk/.platformio/lib/Nanopb/generator/proto'
cd "src" ; \
protoc --plugin=protoc-gen-nanopb=/home/dirk/.platformio/lib/Nanopb/generator/protoc-gen-nanopb \
	--nanopb_out=./ \
	rfquack.proto \
	--python_out=client/
/home/dirk/.platformio/lib/Nanopb/generator/nanopb_generator.py:1619: DeprecationWarning: 'U' mode is deprecated
  Globals.separate_options = read_options_file(open(optfilename, "rU"))

[dirk@t520 RFQuack]$ platformio lib -g install file://./
Library Storage: /home/dirk/.platformio/lib
Library Manager: Installing file://./
Library Manager: RFQuack @ 0.0.1 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing git+https://github.com/nanopb/nanopb.git
git version 2.26.2
Cloning into '/home/dirk/.platformio/.cache/tmp/pkg-installing-fux2zchn'...
remote: Enumerating objects: 616, done.
remote: Counting objects: 100% (616/616), done.
remote: Compressing objects: 100% (539/539), done.
remote: Total 616 (delta 60), reused 404 (delta 31), pack-reused 0
Receiving objects: 100% (616/616), 1.11 MiB | 357.00 KiB/s, done.
Resolving deltas: 100% (60/60), done.
Library Manager: Nanopb @ 0.4.3+sha.a162fa4 has been installed!
Library Manager: Installing TinyGSM @ ~0.6.0
Library Manager: TinyGSM @ 0.6.2 has been installed!
Library Manager: Installing MQTT @ 2.4.7
Library Manager: Warning! More than one package has been found by MQTT @ 2.4.7 requirements:
 - 256dpi/MQTT @ 2.4.7
 - mbed-mqtt/MQTT @ 0.0.0+sha.9cff7b6bbd01
 - mbed-ethernetinterfacew5500-makers/MQTT @ 0.0.0+sha.f53a95f9c1d8
 - mbed-zhangyx/MQTT @ 0.0.0+sha.6de46e8293d9
 - mbed-jwende/MQTT @ 0.0.0-alpha+sha.ca855d2954
 - mbed-edutech/MQTT @ 0.0.0+sha.05fe2e37b93d
 - mbed-ublox/MQTT @ 0.0.0+sha.f313853d22a6
 - mbed-mfiore/MQTT @ 0.0.0+sha.66fba174d875
 - mbed-markl/MQTT @ 0.0.0+sha.150b06e72bf8
 - mbed-nuvoton/MQTT @ 0.0.0+sha.50950f0d9c96
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Library Manager: MQTT @ 2.4.7 has been installed!
Library Manager: Installing Queue @ ~1.6
Library Manager: Warning! More than one package has been found by Queue @ ~1.6 requirements:
 - smfsw/Queue @ 1.8
 - mbed-wbasser/Queue @ 0.0.0+sha.a03810d46457
 - mbed-danilob/queue @ 0.0.0+sha.8b473c4a0afc
 - mbed-smeshlink/Queue @ 0.0.0+sha.abfd5885f80d
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Library Manager: Queue @ 1.6.0 has been installed!
Library Manager: Installing ArduinoLog @ ~1.0.3
Library Manager: ArduinoLog @ 1.0.3 has been installed!
Library Manager: Installing git+https://github.com/rfquack/RadioLib
git version 2.26.2
Cloning into '/home/dirk/.platformio/.cache/tmp/pkg-installing-97ludibs'...
remote: Enumerating objects: 299, done.
remote: Counting objects: 100% (299/299), done.
remote: Compressing objects: 100% (227/227), done.
remote: Total 299 (delta 88), reused 141 (delta 23), pack-reused 0
Receiving objects: 100% (299/299), 581.12 KiB | 1.69 MiB/s, done.
Resolving deltas: 100% (88/88), done.
Library Manager: RadioLib @ 4.0.1+sha.7535faa has been installed!
Library Manager: Installing git+https://github.com/phretor/Base64
git version 2.26.2
Cloning into '/home/dirk/.platformio/.cache/tmp/pkg-installing-63cgrk5e'...
remote: Enumerating objects: 14, done.
remote: Counting objects: 100% (14/14), done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 14 (delta 1), reused 11 (delta 1), pack-reused 0
Receiving objects: 100% (14/14), 13.28 KiB | 13.28 MiB/s, done.
Resolving deltas: 100% (1/1), done.
Library Manager: Base64 @ 0.0.1+sha.925def4 has been installed!
```

dgeppert, 2020-10-12:
however a successive pio run still results in the same kind of errors

[piorun002.log](https://github.com/rfquack/RFQuack/files/5366255/piorun002.log)

however, this one kind of looks strange to me. any idea?

```
/home/dirk/.platformio/lib/RFQuack/src/radio/../modules/../rfquack.pb.h:10:2: error: #error Regenerate this file with the current version of nanopb generator.
 #error Regenerate this file with the current version of nanopb generator.
```

phretor, 2020-10-12:
Oh, I get what's happening! You clone the repo, you generate the .h file from the .proto by running `make proto`, and the `pio install` ruins it all, because it re-clones the repo, so overwriting the *.h file with the one that ships with the repo, which has been generated by a different protobuf compiler.

Can you verify this?

dgeppert, 2020-10-12:
make proto generates new `src/rfquack.pb.h` and `src/rfquack.pb.c` (from the file creating time)

after I am not sure what `platformio lib -g install file://./` does with these files

the example compilation `pio run` finds and complains about `.platformio/lib/RFQuack/src/radio/../modules/../rfquack.pb.h`
so it seems the global library install copied the file over

it also clones the following from git:
```
git+https://github.com/nanopb/nanopb.git
git+https://github.com/rfquack/RadioLib
git+https://github.com/phretor/Base64
```

but it seems it does not clone again, but copies from the local directory:
`Library Manager: Installing file://./`

are you able to reproduce this build problem if you perform the same steps, i.e. reinstall platformio, checkout RFQuack, install the RFQuack libraries into the platformio and the try to build an example?

dgeppert, 2020-10-12:
for what concerns the proto error messages:

before `make proto` we installed:

`platformio lib -g install git+https://github.com/nanopb/nanopb.git#0.3.9.2`

but later in `library.json` the `platformio lib -g install file://./` will install the current master:

```
"name": "Nanopb",
"version": "https://github.com/nanopb/nanopb.git"
```
 
there are also some references to `github.com/trendmicro` which probably should be replace by `github.com/rfquack`

now the error messages are much less, but from the error messages it seems to me that code development advanced and not everything for all classes was consistent, e.g. the missing _br for CC1101 and some more missing (inherited) methods.

so maybe a consistent version of packages would make it possible to compile the examples.

phretor, 2020-10-13:
@dgeppert correct - all the refs should be updated from "trendmicro/rfquack" to "rfquack/rfquack". Good move to upgrade to nanopb@master (someone should definitely update the instructions, but time lacks on my side: feel free to send a pull request if you have time, or get someone involved to help building some community around this tool).

Do you mind posting the output of the building process after you upgraded to nanopb@master? You mentioned you get way fewer error messages, but still getting some: I'm curious to see those!

dgeppert, 2020-10-13:
here the requested log:

```
pio run -e ESP32 
Processing ESP32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
--------------------------------------------------------------------------------
Platform Manager: Installing espressif32
Platform Manager: espressif32 @ 2.0.0 has been installed!
The platform 'espressif32' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing platformio/toolchain-xtensa32 @ ~2.50200.0
Tool Manager: toolchain-xtensa32 @ 2.50200.80 has been installed!
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.10004.191002
Tool Manager: framework-arduinoespressif32 @ 3.10004.200129 has been installed!
Tool Manager: Installing platformio/tool-esptoolpy @ ~1.20600.0
Tool Manager: tool-esptoolpy @ 1.20600.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40001.0
Tool Manager: tool-scons @ 4.40001.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <RFQuack> 0.0.1
|   |-- <Nanopb> 0.4.3+sha.a162fa4
|   |-- <TinyGSM> 0.6.2
|   |-- <MQTT> 2.4.7
|   |-- <Queue> 1.6.0
|   |-- <ArduinoLog> 1.0.3
|   |-- <RadioLib> 4.0.1+sha.7535faa
|   |   |-- <SPI> 1.0
|   |-- <Base64> 0.0.1+sha.925def4
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
Building in release mode
Compiling .pio/build/ESP32/src/main.cpp.o
Generating partitions .pio/build/ESP32/partitions.bin
Compiling .pio/build/ESP32/libd20/Nanopb/pb_common.c.o
Compiling .pio/build/ESP32/libd20/Nanopb/pb_decode.c.o
Compiling .pio/build/ESP32/libd20/Nanopb/pb_encode.c.o
In file included from /home/dirk/.platformio/lib/RFQuack/src/radio/RadioLibWrapper.h:16:0,
                 from /home/dirk/.platformio/lib/RFQuack/src/radio/RFQCC1101.h:5,
                 from /home/dirk/.platformio/lib/RFQuack/src/radio/drivers.h:4,
                 from src/main.cpp:29:
/home/dirk/.platformio/lib/RadioLib/src/RadioLib.h:50:4: warning: #warning "God mode active, I hope it was intentional. Buckle up, lads." [-Wcpp]
   #warning "God mode active, I hope it was intentional. Buckle up, lads."
    ^
Compiling .pio/build/ESP32/libd20/Nanopb/spm-test/objc/c-header.c.o
Archiving .pio/build/ESP32/libae0/libTinyGSM.a
Indexing .pio/build/ESP32/libae0/libTinyGSM.a
Compiling .pio/build/ESP32/libfb7/MQTT/MQTTClient.cpp.o
Archiving .pio/build/ESP32/libd20/libNanopb.a
Compiling .pio/build/ESP32/libfb7/MQTT/lwmqtt/client.c.o
Indexing .pio/build/ESP32/libd20/libNanopb.a
Compiling .pio/build/ESP32/libfb7/MQTT/lwmqtt/helpers.c.o
In file included from /home/dirk/.platformio/lib/RadioLib/src/TypeDef.h:4:0,
                 from /home/dirk/.platformio/lib/RadioLib/src/RadioLib.h:43,
                 from /home/dirk/.platformio/lib/RFQuack/src/radio/RadioLibWrapper.h:16,
                 from /home/dirk/.platformio/lib/RFQuack/src/radio/RFQCC1101.h:5,
                 from /home/dirk/.platformio/lib/RFQuack/src/radio/drivers.h:4,
[... 19 lines cut ...]
/home/dirk/.platformio/lib/RFQuack/src/radio/RFQnRF24.h:67:13: error: 'int16_t RFQnRF24::setSyncWord(uint8_t*, pb_size_t)' marked 'override', but does not override
     int16_t setSyncWord(uint8_t *bytes, pb_size_t size) override {
             ^
src/main.cpp:33:35: error: 'NC' was not declared in this scope
 RadioA radioA = new Module(2, 21, NC);
                                   ^
Compiling .pio/build/ESP32/libfb7/MQTT/lwmqtt/packet.c.o
Compiling .pio/build/ESP32/libfb7/MQTT/lwmqtt/string.c.o
Compiling .pio/build/ESP32/lib716/Queue/cppQueue.cpp.o
*** [.pio/build/ESP32/src/main.cpp.o] Error 1
Compiling .pio/build/ESP32/lib6ea/ArduinoLog/ArduinoLog.cpp.o
========================= [FAILED] Took 34.98 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
ESP32          FAILED    00:00:34.981
==================== 1 failed, 0 succeeded in 00:00:34.981 ====================
```

- undefining `RADIOLIB_DEBUG` works-around the first `pragma` error message
- the virtual overrides e.g. setSyncWord may be edited by hand (`pb_size_t` unknown use `uint8_t` as a work-around)

the same kind of errors show up in successivly compiling other radio modules, the main problem seems the private bitrate variable `_br`

hth & cheers

phretor, 2020-10-13:
Spot on!

https://github.com/rfquack/RFQuack/blob/master/src/radio/RFQCC1101.h#L205

@Guglio95 can you check whether you forgot to commit some changes? That looks to be part of [RadioLib](https://github.com/jgromes/RadioLib) - so maybe it's a fork that we forgot to PR to RadioLib?

FiorixF1, 2020-11-28:
Hi everybody!

I tried to compile from source and diagnosticate the errors. I succeeded in finding the cause and fixing them. Basically there are some inconsistencies in (our version of) RadioLib:

* ```NC``` is no more defined and it is replaced by ```RADIOLIB_NC```
* ```_br``` is undefined in CC1101 but it is used later, so we have to define it
* Some code related to CC1101 is mysteriously inside the class of RF69, causing compiling issues

I corrected these problems and now it works flawlessly by my side. FYI I compiled and flashed using ```platformio run --target upload``` instead of pio. I'm going to make a PR on both RFQuack and RadioLib.

dgeppert, 2020-11-28:
Hello Alessandro @FiorixF1,

many thanks for your efforts bringing the repos in consistent states again.

I had to look up the different commands I used to build:

```
rm -rf ~/.platformio/
cd platformio/
python get-platformio.py
sudo ln -sf ~/.platformio/penv/bin/platformio /usr/local/bin/platformio
sudo ln -sf ~/.platformio/penv/bin/pio /usr/local/bin/pio
sudo ln -sf ~/.platformio/penv/bin/piodebuggdb /usr/local/bin/piodebuggdb
platformio lib -g install git+https://github.com/nanopb/nanopb.git # match version in RFQuack/library.json
cd ../
rm -rf RFQuack
git clone https://github.com/rfquack/RFQuack.git
cd RFQuack
make proto
platformio lib -g install file://./
cd examples/RFQuack-esp32-CC1101-serial/
platformio run --target upload
```
However, I still get error messages in the build of the CC1101 example, please see the attached log file:

[RFQuack_compile_run_20201128a.txt](https://github.com/rfquack/RFQuack/files/5611062/RFQuack_compile_run_20201128a.txt)

If it works for you, what are you doing different (and I am doing wrong ;-) )

Best regards, Dirk

FiorixF1, 2020-11-28:
Sure.

I modified this repository, but also [RFQuack-cli](https://github.com/rfquack/RFQuack-cli) and [RadioLib](https://github.com/rfquack/RadioLib). My PR here has been accepted, but those in the other two repos not yet. So I ask @phretor or @Guglio95 if they can accept them, since they are necessary to fix the errors. In the meantime, you could use my fork [here](https://github.com/FiorixF1/RFQuack-cli) for the CLI, and change "https://github.com/rfquack/RadioLib" in "https://github.com/FiorixF1/RadioLib" inside ```library.json``` so that it fetches the patched version of RadioLib.

To compile the examples, I copied ```platformio.ini``` into the root directory, and ```main.cpp``` into ```/src/``` next to the other source files. Then I run ```platformio run --target upload``` from the root directory.

If the compilation and flashing are successful, you should be able to communicate with the dongle using the CLI with the command ```rfquack -l DEBUG tty --port /dev/ttyUSB0``` (change port with the actual one if necessary).

dgeppert, 2020-11-29:
Hello Alessandro @FiorixF1,

yes, this looks so much better!!!

using your RadioLib fork I could successfully compile and upload to my ESP32.

then using your RFQuack-cli (and fiddling around to set up a penv for Python 3.8) I could get into RFQuack:

```
                   ,-.
               ,--' ~.).
             ,'         `.
            ; (((__   __)))      welcome to rfquack!
            ;  ( (#) ( (#)
            |   \_/___\_/|              the versatile
           ,"  ,-'    `__".             rf-hacking tool that quacks!
          (   ( ._   ____`.)--._        _
           `._ `-.`-' \(`-'  _  `-. _,-' `-/`.
            ,')   `.`._))  ,' `.   `.  ,','  ;   ~~~
          .'  .     `--'  /     ).   `.      ;
         ;     `-        /     '  )         ;           ~~~~
         \                       ')       ,'    ~~  ~
          \                     ,'       ;           ~~
           \               `~~~'       ,'               ~~~  ~~    ~~~~~
            `.                      _,'             ~~~
        hjw   `.                ,--'
        ~~~~~~~~`-._________,--'~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

        ------------------------------------------------------------------------
```
However now I need to check the cabling between the ESP devkit4 (probably much different from the older Arduino Feather / devkit1) and the CC1101 module since the communication does not seem to work. But that's a project for another day ;-)

btw a small comment (no complain): RFQuack Readme.md contains some photos of the modules but not the cabling. Maybe if familiar with the Ardafruit playground this is obvious. I had to find the module descriptions and and kind-of reverse-engineer to make a guess how it could be cabled to adapt it to my actual ESP.

Many thanks to the team for your help getting this compiled and run. A fun project! Maybe I will try to make it talk to the Yardstick One which I bought in the meantime.

Mille grazie e a piu tardi!

P.S. I do not close the issue until Alessandro's fixes make it into the corresponding project masters, but of course you may just close it if you think it's been solved.

phretor, 2020-12-08:
Also relevant if you're using CC1101 - https://github.com/jgromes/RadioLib/issues/208

TheChatty, 2021-04-11:
@dgeppert: Have you figured out the correct cabling?

@phretor: I also got the ESP32_Devkitc_V4 and even two CC1101 (one optimized for 433Mhz the other for 868MHz). Could both transceivers be connected to my esp32? Could you please provide a reference cabling along with corresponding init commands? Have all repos been aligned already?

How does this project compare to [SignalDuino](https://github.com/RFD-FHEM/SIGNALDuino) (running on an ESP32 it also utilizes a CC1101 to decode yet unknown signals) in terms of visibility of unknown signals? I've got a [868MHz doorbell](https://github.com/RFD-FHEM/RFFHEM/issues/946) unfortunately not visible by the former project. Similarily with [Somfy protocol](https://github.com/RFD-FHEM/SIGNALDuino/issues/160).

dgeppert, 2021-04-11:
hello @TheChatty 

I connected the CC1101 to az-delivery-devkit-v4 VSPI bus, see the [pinout diagram](https://www.az-delivery.de/products/esp-32-dev-kit-c-v4) 

 ```
CC1101  ESP32
  Pin      
 1  VCC  3V
 2  Gnd  Gnd
 3  SI   GPIO23/MOSI
 4  SCLK GPIO18/CLK
 5  SO   GPIO19/MISO
 6  GD2  GPIO21 (unused)
 7  GD0  GPIO4
 8  CSN  GPIO5/SS
```
depending on if you are using Docker build/run or one of the examples:
```
docker/project/platformio.ini
examples/RFQuack-esp32-CC1101-serial/platformio.ini
env      board 
ESP32 	 az-delivery-devkit-v4
```

```
examples/RFQuack-esp32-CC1101-serial/src/main.cpp : 
RadioA radioA = new Module(5, 4, RADIOLIB_NC); // CS IRQ
```

`docker run ... -e BOARD=ESP32 -e RADIOA=CC1101 -e RADIOA_CS=5 -e RADIOA_IRQ=4  ...`

however, I am using RadioLib and RFQuack from @FiorixF1 because I experienced compilation issues.

and somhow my CC1101 uses the legacy `CC1101_CHIP_VERSION 0x04` instead of the standard `0x14`, so `RadioLib/src/modules/CC1101/CC1101.h and CC1101.cpp` were adjusted.

hth

phretor, 2021-04-12:
Just checking if this issue needs any other actions or we could close it.

Please check the new Dockerfile (I just committed here) and respective images I just pushed (to Docker Hub).

dgeppert, 2021-04-13:
Hello Fede,

great to see you aligned to RadioLib and all compilation issues disappeared.

I just ran the latest docker hub images for rfquack and rfquack-cli and also built local ones, all ran and compiled well for ESP32 and CC1101.

Great to see this progress and efforts, now one may actually start the interesting stuff: using RFQuack!

mille grazie

closed 2021-04-13 by dgeppert

TheChatty, 2021-04-14:
@dgeppert: Thank you so much for helping out. You saved myself a lot of time for research. When I finally started the CLI along with the baud parameters I could eventually use q.guessing.start() to see non-chunked output while pressing some remote control buttons. May the quacking begin.

HomeAutoUser, 2021-06-03:
@TheChatty 
Are there any successes with the doorbell?

TheChatty, 2021-06-04:
With RFQuack I could only see receiptions from remotes which also worked with SIGNALDuino (all with 433 MHz). My doorbell (868 MHz) was not visible (I changed the frequency of course).

But I found the general usage of RFQuack quite unintuitive and not well documented thus I gave up.

phretor, 2021-06-04:
> But I found the general usage of RFQuack quite unintuitive and not well documented thus I gave up.

Can't agree more 😄 RFQuack is meant to be generic rather than a tool that works out of the box, so the learning curve is definitely steeper.

Documentation may really help, but my time (like everyone else, I'd say) is very limited. If anyone is feeling to contribute with some doc improvements, PRs are more than welcome! 🍻