rfquack/RFQuack #4: Using HUZZAH32 w/RFM69HCW

dgeppert, 2020-12-04:
Hello,

to start with I ordered and assembled the same Adafruit modules found on the photos on the RFQuack github site:

[image]

Since there is not much one may do wrong there I thought it should be straightforward to get at least some communication between the HUZZAH32 ESP and the RFM69HCW CC1101 Radio Feather (I used the same wiring for the for the CC1101 CS, IRQ and RST as one may find on the github photos).

However there is no communication on my modules (I programmed the HUZZAH32 with some other examples using Arduino IDE and Espressif-IDE and the module seems fine).

From the photos and the schematics, I used Radio CS=13, IRQ=27 i.e. Radio = new Module(13, 27, RADIOLIB_NC). By chance, since this is exactly the same setup: can anybody confirm these? 

To me this seems quite strange: these Adafruit Feathers use GPIO 18=MOSI, 19=MISO, 5=SCK whereas the ESP32 VSPI is on GPIO 23=MOSI, 19=MISO, 18=SCLK. Of course one may use whatever IO ports and implement a SPI bus, but it would be better to use the one provided by the hardware chip. Anyway this seems just a detail and probably not the cause for my problem.
 
cheers, Dirk

dgeppert, 2020-12-05:
Hello,

following some examples in [(https://learn.adafruit.com/radio-featherwing/using-the-rfm69-radio] I could establish some SPI  communication between the HUZZAH32 and the RFM69 Radio. So at least my hardware and cabling seem fine. 

@FiorixF1 : when pulling your RadioLib again today I noticed you made some progress : [RF69] Fixed setOOK()
however this results in some compilation error. maybe:

`int16_t state = SPIsetRegValue(RF69_REG_DATA_MODUL, RF69_OOK, 4, 3);`

should read:

`int16_t state = _mod->SPIsetRegValue(RF69_REG_DATA_MODUL, RF69_OOK, 4, 3);`

with the two lines modified compilation is fine, mut not sure if it's what you intend, there is also a function with the same name defined in Module.h/cpp.

P.S. nice to see there is some progress ;-)

dgeppert, 2020-12-05:
switching on RADIOLIB_DEBUG and monitoring the tty via screen:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[RFQK]        159 T: Standalone mode network setup: done!
[RF[RFQK]        485 T: Serial transport connected
[RFQK]        585 T: Serial transport connected
R       71       0
CC1101 not found! (1 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
R       71      0
CC1101 not found! (2 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
R       71      0
CC1101 not found! (3 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
R       71      0
CC1101 not found! (4 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
R       71      0
CC1101 not found! (5 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
R       71      0
CC1101 not found! (6 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
R       71      0
CC1101 not found! (7 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
R       71      0
CC1101 not found! (8 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
R       71      0
CC1101 not found! (9 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
R       71      0
CC1101 not found! (10 of 10 tries) CC1101_REG_VERSION == 0x0000, expected 0x0014
No CC1101 found!
[RFQK]        786 E: Unable to initialize radio, got code -2
[RFQK]        786 T: Something went wrong, check your wiring.
```

phretor, 2020-12-05:
Alrigh @dgeppert - this shows that the module is configured to use a CC1101, which is of course not found. Can you share (or point us) to the full `main` file? I assume you're using a RFM69HCW, which will require some work on your side as there needs to be the equivalent of this [RFQCC1101.h](https://github.com/rfquack/RFQuack/blob/master/src/radio/RFQCC1101.h) but for the RFM69HCW.

It's not a lot of work because the RFQCC1101.h is just a thin abstraction layer between RFQuack and RadioLib.

dgeppert, 2020-12-05:
for some reason I had in mind that this is just a CC1101, some strange conclusion from the photos and the example sources. my stupid fault ;-)

so I am just using **RFQuack/examples/RFQuack-esp32-CC1101-serial**

```
*****************************************************************************
 * RFQuack configuration/
 *****************************************************************************/

/* ID definition */
#define RFQUACK_UNIQ_ID "ESP32_CC1101"
#define RFQUACK_TOPIC_PREFIX "MAKE_THIS_UNIQUE"

/* Transport configuration */
#define RFQUACK_TRANSPORT_SERIAL
#define RFQUACK_SERIAL_BAUD_RATE 115200

/* Enable Radio debug messages */
#define RFQUACK_LOG_ENABLED
#define RFQUACK_DEV
#define RADIOLIB_DEBUG

/* Disable Software Serial logging */
#define RFQUACK_LOG_SS_DISABLED

/* Default radio config */
// carrier frequency:                   868.0 MHz
// bit rate:                            4.8 kbps
// Rx bandwidth:                        325.0 kHz
// frequency deviation:                 48.0 kHz
// sync word:                           0xD391

/* Radio configuration */
#include <radio/drivers.h>

#define USE_RADIOA
typedef RFQCC1101 RadioA;
RadioA radioA = new Module(33, 27, RADIOLIB_NC);

// Uncomment to add a new radio, then change setup(){ rfquack_setup(radioA, radioB); }
// #define USE_RADIOB
// typedef RFQCC1101 RadioB;
// RadioB radioB = new Module(3, 22, RADIOLIB_NC);

/*****************************************************************************
 * /RFQuack configuration
 *****************************************************************************/
#include <rfquack.h>

void setup() {
  rfquack_setup(&radioA);
}

 void loop() {
   rfquack_loop();
 }
```
so probably just `typedef RFQCC1101 RadioA;` to change, but for what?

By chance: will this RFM69HCW be able to communicate with a CC1101?

dgeppert, 2020-12-05:
that said, I replaced the Adafruit Feather RFM69 with a CC1101 module.

somehow MISO and MOSI are swapped and the CHIP_VERSION identifier for my chip is 0x04 instead of the 0x14 in the CC1101 datasheet (may be changed without notice!). allowing this id in the RadioLib CC1101, the modul is recongnised and RFQuack-cli is able to detect and communicate with it (e.g. read out registers). 

next step is to communicate with a YardStick, make the RFM69 work and try a non-Feather ESP.

phretor, 2021-04-13:
Is this issue still happening with the most recent commits @dgeppert ?

dgeppert, 2021-04-13:
hello @phretor 

giving a `docker run` using the RFQuack hub image (as well as a recent local image) results in the compilation/link errors below.

btw I noticed that the `Readme.md` still says `nanopb/nanopb.git#0.3.9.2` whereas the library.json just installs/overwrites it with `nanopb/nanopb.git` (so no need to install it before with the older version)

hth

```
Huzzah32 and RF69
-----------------

build.env
  BOARD=HUZZAH32
  RADIOA=RF69
  RADIOA_CS=13
  RADIOA_IRQ=27
  RADIOA_CE=15

using RFQuack Docker hub image
docker run --device /dev/ttyUSB0:/board -e BOARD=HUZZAH32 -e RADIOA=RF69   -e RADIOA_CS=13 -e RADIOA_IRQ=27 -e RADIOA_CE=15 -l DEBUG --rm -it rfquack/rfquack

[dirk@t520 RFQuack_20210413]$ docker images --all
REPOSITORY          TAG                 IMAGE ID            CREATED             SIZE
[dirk@t520 RFQuack_20210413]$ docker run --device /dev/ttyUSB0:/board -e BOARD=HUZZAH32 -e RADIOA=RF69   -e RADIOA_CS=13 -e RADIOA_IRQ=27 -e RADIOA_CE=15 -l DEBUG --rm -it rfquack/rfquack
Unable to find image 'rfquack/rfquack:latest' locally
latest: Pulling from rfquack/rfquack
f7ec5a41d630: Pull complete 
388aab6f3954: Pull complete 
9501c65d3a7f: Pull complete 
ac4bddc73958: Pull complete 
db3cd99a2d05: Pull complete 
29bb9020e4c3: Pull complete 
27522cf07cc5: Pull complete 
32a954c0ffdb: Pull complete 
a4b23abb575a: Pull complete 
8c1c281de93c: Pull complete 
3bba3e45d66a: Pull complete 
59947e610fa2: Pull complete 
Digest: sha256:1e62155a58ed5f9660145eea60eb39edb613e5d0592327682fff444ee84eabd7
Status: Downloaded newer image for rfquack/rfquack:latest
Installation ID is 69e21d5a-6daf-4158-a374-6ed2f5142597
Processing HUZZAH32 (platform: espressif32; board: featheresp32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/framework-arduinoespressif32 @ ~3.10006.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-arduinoespressif32 @ 3.10006.210326 has been installed!
Tool Manager: Installing platformio/tool-mkspiffs @ ~2.230.0
Tool Manager: tool-mkspiffs @ 2.230.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40100.2
Tool Manager: tool-scons @ 4.40100.2 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32.html
PLATFORM: Espressif 32 (3.2.0) > Adafruit ESP32 Feather
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <RFQuack> 0.0.1
|   |-- <Nanopb> 0.4.5+sha.29b8ee4
|   |-- <TinyGSM> 0.6.2
[... 146 lines cut ...]
Compiling .pio/build/HUZZAH32/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/main.cpp.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/HUZZAH32/libFrameworkArduino.a
Linking .pio/build/HUZZAH32/firmware.elf
.pio/build/HUZZAH32/src/main.cpp.o:(.literal._ZN7RFQRF6913writeRegisterEthhh[RFQRF69::writeRegister(unsigned short, unsigned char, unsigned char, unsigned char)]+0x0): undefined reference to `RF69::SPIsetRegValue(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'
.pio/build/HUZZAH32/src/main.cpp.o: In function `RFQRF69::writeRegister(unsigned short, unsigned char, unsigned char, unsigned char)':
/quack/src/main.cpp:57: undefined reference to `RF69::SPIsetRegValue(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'
.pio/build/HUZZAH32/src/main.cpp.o: In function `RFQRF69::begin()':
/quack/src/main.cpp:57: undefined reference to `RF69::SPIsetRegValue(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
*** [.pio/build/HUZZAH32/firmware.elf] Error 1
============================================================================== [FAILED] Took 59.02 seconds ==============================================================================

Environment    Status    Duration
-------------  --------  ------------
HUZZAH32       FAILED    00:00:59.022
========================================================================= 1 failed, 0 succeeded in 00:00:59.022 =========================================================================
```

phretor, 2021-04-14:
Spotted the issue, https://github.com/rfquack/RFQuack/commit/master#diff-680c089ca24756366139fdad37516f4e3b0f5780df2b692a09cd282421111bc4R25

Please try now. Currently pushing Docker images, but you can still try locally with:

```bash
$ make docker-build-nc   # <- drop this if you have the latest Docker image (in the future).
$ make build
```

dgeppert, 2021-04-14:
tried Docker image and local build, both with same result...

something missing on my side?

```
[dirk@t520 RFQuack_20210414]$ docker run --device /dev/ttyUSB1:/board -e BOARD=HUZZAH32 -e RADIOA=RF69   -e RADIOA_CS=13 -e RADIOA_IRQ=27 -e RADIOA_CE=15 -l DEBUG --rm -it rfquack/rfquack
Unable to find image 'rfquack/rfquack:latest' locally
latest: Pulling from rfquack/rfquack
f7ec5a41d630: Already exists 
388aab6f3954: Already exists 
9501c65d3a7f: Already exists 
ac4bddc73958: Already exists 
db3cd99a2d05: Already exists 
4f4fb700ef54: Pull complete 
0ab1417cfdf3: Pull complete 
c978d3c816fa: Pull complete 
01cefccd3c2a: Pull complete 
3e7503b2ba43: Pull complete 
367f94a8accc: Pull complete 
6c47fb2568c2: Pull complete 
c7b3521ab958: Pull complete 
Digest: sha256:f6f67c64fdd75ee5b64a0e0847692b23807eb92a3f517aceced2fb302742005f
Status: Downloaded newer image for rfquack/rfquack:latest
Installation ID is 87824cea-9b10-408b-bdaf-c6c7e548b119
Processing HUZZAH32 (platform: espressif32; board: featheresp32; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
...
...
...
Linking .pio/build/HUZZAH32/firmware.elf
.pio/build/HUZZAH32/src/main.cpp.o:(.literal._ZN7RFQRF6913writeRegisterEthhh[RFQRF69::writeRegister(unsigned short, unsigned char, unsigned char, unsigned char)]+0x0): undefined reference to `RF69::SPIsetRegValue(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'
.pio/build/HUZZAH32/src/main.cpp.o: In function `RFQRF69::writeRegister(unsigned short, unsigned char, unsigned char, unsigned char)':
/quack/src/main.cpp:57: undefined reference to `RF69::SPIsetRegValue(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
*** [.pio/build/HUZZAH32/firmware.elf] Error 1
============================================================================== [FAILED] Took 62.45 seconds ==============================================================================

Environment    Status    Duration
-------------  --------  ------------
HUZZAH32       FAILED    00:01:02.445
========================================================================= 1 failed, 0 succeeded in 00:01:02.445 =========================================================================

```

phretor, 2021-04-14:
I assume you pulled the latest commit, correct?

dgeppert, 2021-04-14:
actually I just noticed that you also enabled god mode in Radiolib and it was using maybe my installed version, so I cleaned platformio/lib and tried again. 

```
git clone https://github.com/rfquack/RFQuack.git
cd RFQuack

rm -rf ~/.platformio/lib
make proto
platformio lib -g install file://./
docker rmi rfquack/rfquack

vi build.env
make docker-build-nc
make build
```

```
...
Linking .pio/build/HUZZAH32/firmware.elf
.pio/build/HUZZAH32/src/main.cpp.o:(.literal._ZN7RFQRF6913writeRegisterEthhh[RFQRF69::writeRegister(unsigned short, unsigned char, unsigned char, unsigned char)]+0x0): undefined reference to `RF69::SPIsetRegValue(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'
.pio/build/HUZZAH32/src/main.cpp.o: In function `RFQRF69::writeRegister(unsigned short, unsigned char, unsigned char, unsigned char)':
/quack/src/main.cpp:57: undefined reference to `RF69::SPIsetRegValue(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)'
collect2: error: ld returned 1 exit status
*** [.pio/build/HUZZAH32/firmware.elf] Error 1
============================================================================== [FAILED] Took 58.20 seconds ==============================================================================

Environment    Status    Duration
-------------  --------  ------------
HUZZAH32       FAILED    00:00:58.199
========================================================================= 1 failed, 0 succeeded in 00:00:58.199 =========================================================================
```

dgeppert, 2021-04-16:
hello,

setting up an example `RFQuack-esp32-RF69-serial` in order to use a local non-docker and investigating the linking issue:

```
RFQRF69::writeRegister(unsigned short, unsigned char, unsigned char, unsigned char)':
undefined reference to RF69::SPIsetRegValue(unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)
```

changed:

```
RFQuack/RadioLib/src/radio/RFQRFM69.h
    writeRegister(rfquack_register_address_t reg, rfquack_register_value_t value, uint8_t msb, uint8_t lsb) override {
      //SPIsetRegValue((uint8_t) reg, (uint8_t) value, msb, lsb, 0);
      _mod->SPIsetRegValue((uint8_t) reg, (uint8_t) value, msb, lsb, 0); // ---> linking OK 
```

As further up, using `_mod `when calling `SPIsetRegValue` and linking/flushing is ok.

However, RFQuack-cli does not connect to the dongle and monitoring the serial usb shows:

```
[RFQK]        131 T: Standalone mode network setup: done!
[RFQK]        455 T: Serial transport connected
[RFQK]        556 T: Serial transport connected
[RFQK]        671 E: setFrequencyDeviation was not implemented.
[RFQK]        671 E: Unable to initialize radio, got code -590
[RFQK]        672 T: Something went wrong, check your wiring.
```
se checking my setup/wiring...

phretor, 2021-04-16:
Thanks for insisting @dgeppert - I had fixed only one of the two calls to `SPIsetRegValue` and not the on at L103 :-)

phretor, 2021-04-16:
BTW, some of the methods are still to be implemented, but given RadioLib's abstraction, it shouldn't be too much coding. I'll implement them next week, but if anyone has time to contribute, they'll be greatly appreciated.

File: `src/radio/RFQRFM69.h` (notice the `TODO`)

```
    int16_t jamMode() {
      Log.error(F("TODO: jamMode was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t setPreambleLength(uint32_t size) {
      Log.error(F("TODO setPreambleLength was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t setFrequencyDeviation(float freqDev) {
      Log.error(F("TODO setFrequencyDeviation was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }
    
    int16_t getFrequencyDeviation(float &freqDev) {
      Log.error(F("TODO getFrequencyDeviation was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t setRxBandwidth(float rxBw) {
      Log.error(F("TODO setRxBandwidth was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t setOutputPower(uint32_t txPower) {
      Log.error(F("TODO: setOutputPower was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t fixedPacketLengthMode(uint8_t len) {
      Log.error(F("TODO fixedPacketLengthMode was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t variablePacketLengthMode(uint8_t len) {
      Log.error(F("TODO variablePacketLengthMode was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t setPromiscuousMode(bool isPromiscuous) {
      Log.error(F("TODO setPromiscuousMode was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t setCrcFiltering(bool crcOn) {
      Log.error(F("TODO setCrcFiltering was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t setAutoAck(bool autoAckOn) {
      Log.error(F("TODO setAutoAck was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t isCarrierDetected(bool &isDetected) {
      Log.error(F("TODO isCarrierDetected was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    float getRSSI(float &rssi) {
      Log.error(F("TODO getRSSI was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }

    int16_t setModulation(rfquack_Modulation modulation) {
      Log.error(F("TODO setModulation was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }
    
    int16_t getModulation(char *modulation) {
      Log.error(F("TODO getModulation was not implemented."));
      return ERR_COMMAND_NOT_IMPLEMENTED;
    }
```

Example implementations for the CC1101: `src/radio/RFQCC1101.h`

Reference internal methods in RadioLib:
* https://github.com/rfquack/RadioLib/blob/master/src/modules/RF69/RF69.h#L440
* https://github.com/rfquack/RadioLib/blob/master/src/modules/CC1101/CC1101.h#L506

dgeppert, 2021-04-16:
hello,

I tested my RF69 with Arduino IDE and Adafruit RadioHead, got the Init OK message, so wiring should be OK.

I am struggling to get RADIOLIB_DEBUG and RADIOLIB_VERBOSE to work.

would like to see all these debug prints from RadioLib/src/modules/RF69/RF69.cpp

tried: defining it in 
- example/.../src/main.cpp
- Radiolib/src/BuildOpt.h

Maybe you could provide instructions to enable this the correct way? 
Is there a way to switch this on also when running a Docker image or does it has to be build with the proper option before?

Thanks!

phretor, 2021-04-16:
Yes, wiring is OK. Check latest comment on this thread: what's happening is that some methods are only stubs for the RFM69, so they need to be implemented.

About trace-ability: the only way to have them enabled is at pre-processing time (e.g., before the actual compile process) or the source code will grow too much and we'll loose performance substantially. So, the best way to enable them is to `#define` them as early as possible. For example, RadioLib provides a `BuildOpt.h`, which is a good place to work for global settings/variables/macros. That's where, in our own RadioLib fork, we enable the infamous "God mode" :-)

If you want to have these settings enabled in Docker, you'd have to build your own image definition in a separate `Dockerfile`, where you could insert some patches at the right time (i.e., right after RadioLib is pulled from GitHub and after RFQuack is copied from the host to the container being built). You can do this with `sed` and some simple regexes.

My plan is to close this issue once all the methods listed above as "TODO" will be implemented (either by me or someone here.

dgeppert, 2021-04-16:
Hello @phretor 

so no way I get these RADIOLIB_DEBUG message appear on my screen, maybe I understand something wrong. here what I am doing:

- install RFQuack 
```
git clone https://github.com/rfquack/RFQuack.git
cd RFQuack
make proto
platformio lib -g install file://./
```
- now all is installed in `~/.platformio/libs`
- edit `RadioLib/src/BuildOpt.h` to include `#define` for `RADIOLIB_DEBUG` and `RADIOLIB_VERBOSE`
- `RFQuack/examples/RFQuack-esp32-CC1101-serial`
edit `src/main.cpp` to also include these  `#define`
- build and flush example code:
```
pio run -e ESP32
pio run -e ESP32 --target upload
```
- monitor usb port using screen/picomm

however, all I get is:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[RFQK]        132 T: Standalone mode network setup: done!
[RFQK]        457 T: Serial transport connected
[RFQK]        558 T: Serial transport connected
R	71	4	
R	18	4	
W	18	14	
R	18	14	
R	7	4	
W	7	4	
R	7	4	
R	8	45	
W	8	5	
R	8	5	
R	8	5	
W	8	5	
R	8	5	
R	D	1E	
W	D	10	
R	D	10	
R	E	C4	
W	E	B1	
R	E	B1	
R	F	EC	
W	F	3B	
R	F	3B	
R	7E	C6	
W	7E	60	
R	7E	60	
R	10	8C	
W	10	8A	
R	10	8A	
R	11	22	
W	11	E3	
R	11	E3	
R	10	8A	
W	10	AA	
R	10	AA	
R	15	47	
W	15	47	
R	15	47	
R	15	47	
W	15	47	
R	15	47	
R	7E	60	
W	7E	C0	
R	7E	C0	
R	8	5	
W	8	5	
R	8	5	
R	6	FF	
W	6	FF	
R	6	FF	
R	13	22	
W	13	2	
R	13	2	
R	12	2	
W	12	2	
R	12	2	
R	12	2	
W	12	2	
R	12	2	
R	8	5	
W	8	5	
R	8	5	
R	12	2	
W	12	2	
R	12	2	
R	4	D3	
W	4	12	
R	4	12	
R	5	91	
W	5	AD	
R	5	AD	
R	1B	3	
W	1B	43	
R	1B	43	
R	1B	43	
W	1B	7B	
R	1B	7B	
R	8	5	
W	8	5	
R	8	5	
R	7	4	
W	7	0	
R	7	0	
[RFQK]        963 T: Module 'guessing' registered.
[RFQK]        963 T: Module 'frequency_scanner' registered.
[RFQK]        963 T: Module 'mouse_jack' registered.
[RFQK]        966 T: Module 'packet_filter' registered.
[RFQK]        971 T: Module 'packet_modification' registered.
[RFQK]        976 T: Module 'packet_repeater' registered.
[RFQK]        981 T: Module 'roll_jam' registered.
[RFQK]        986 T: Module 'ping' registered.
[RFQK]        990 T: Module 'radioA' registered.
[RFQK]        994 T: Setup is over.
[RFQK]       1008 T: Starting main loop.
```
but, where are the RADIOLIB_DEBUG messages?

thx

dgeppert, 2021-04-16:
ok, apparently I was misunderstanding the RadioLib debug output.

testing with an example from RadioLib and Arduino IDE it became more clear. 

actually all the R/W/T in my previous post are from RADIOLIB_VERBOSE, so it is working well.

sorry for creating confusion... feel free to delete my previous post, not related to the topic anyway

cheers

phretor, 2021-04-23:
@dgeppert I implemented all of the relevant methods listed above, built Docker images and pushed as `rfquack/rfquack:latest`.

Please let me know if it works as expected. Else feel free to re-open this issue.

closed 2021-04-23 by phretor

dgeppert, 2021-04-23:
ciao @phretor 

seems you got quite some work done this week and implemented all these TODO for RF69.

Using Docker Hub, local docker build and example for RF69, all builds and uploads fine.
However I get the same error messages when monitoring for all of them:

```
[RFQK]        131 T: Standalone mode network setup: done!
[RFQK]        456 T: Serial transport connected
[RFQK]        671 E: **setFrequencyDeviation was not implemented**.
[RFQK]        672 E: Unable to initialize radio, got code  -590
[RFQK]        672 T: Something went wrong, check your wiring.
```
When writing a small stub in `~/.platformio/lib/RFQuack/src/radio/RFQRFM69.h `
```
//    using RF69::setFrequencyDeviation;
 ...
    int16_t setFrequencyDeviation(float freqDev) override {
      freqDev = RF69::setFrequencyDeviation(freqDev);
      return ERR_NONE;
    }
```
(probably wrong, but just to see if rfquack-cli stumbles over this one) and building the example, rfquack-cli manages to start up and to communicate with the RF69.

For now I did not succeed to actually capture any packages, will have to fiddle around a little trying different settings.

Note: OOK although an accepted value produces an error...

```
RFQuack(/dev/ttyUSB0)> q.radioA.set_modem_config(modulation="toto")
rfquack.core[1] ERROR modulation must be one of FSK2, OOK
result = 0
message = 0 changes applied and 0 failed.

RFQuack(/dev/ttyUSB0)> q.radioA.set_modem_config(modulation="OOK")
result = -1
message = 0 changes applied and 1 failed.

RFQuack(/dev/ttyUSB0)> q.radioA.set_modem_config(modulation="FSK2")
result = 0
message = 1 changes applied and 0 failed.
```
Well and CC1101 stumbles over the same `**setFrequencyDeviation was not implemented**` issue and rfquack-cli does not talk to the dongle. Probably you tried to clean up some code there too, but now it seems broken, pity ;-( I hope I did not delete my previous git clones for a consistent set of sources!

cheers, Dirk