rfquack/RFQuack #5: Teensy support

dgeppert, 2020-12-04:
Hello,

on the photos one may see a Teensy 3.2 and the Feather Adapter, but later in the code I did not find any reference for the platformio.ini (and the configuration used). Setting it to the standard teensy31 results in compilation errors. 

So my question is: did you use a Teensy for RFQuack and if yes, what was the configuration?

thanks & cheers, Dirk

phretor, 2021-04-13:
I used the Teensy with changes in the `platformio.ini` file, but I had to change some files here and there to make compilation go through. Was long time ago so I don't have the log files. If you have some recent compilation errors, please paste them here.

Would be good to have a `examples/RFQuack-Teensy32-<radio>-serial`, or maybe you already have one and just need to commit? :-)

dgeppert, 2021-04-13:
ciao Fede, 

also a long time I did not touch the Teensy. for now happy with Huzzah32 and ESP32.
I will close this issue and come back if once I'll give it again a try.

a piu tardi

closed 2021-04-13 by dgeppert