rfquack/RFQuack #6: Error tool-mkspiffs

joelsernamoreno, 2021-04-03:
Hello!

When I try to flash the rfquack firmware with docker I get the following error: Error: Detected unknown package 'tool-mkspiffs'

[image]

Can you check this?

Thanks

phretor, 2021-04-03:
Looks new. I suspect it’s due to PlatformIO not finding that package. If you have a PlatformIO installation on the host, can you try to install only that package and see what happens?

joelsernamoreno, 2021-04-03:
Hi!

If I run pio update you can see that tool-mkspiffs is installed correctly:

[image]

Platformio includes this package automatically. If you remove this check from the rfquack installation I think this problem will be solved

joelsernamoreno, 2021-04-03:
I tried flashing RFQuack on my old laptop (it worked here before), but now I get the same error

I think platformio has made some modification and this is not working now

If you remove the tool-mkspiffs installation in rfquack this should work

joelsernamoreno, 2021-04-03:
To manually install tool-mkspiffs you can run:

platformio platform install espressif32 --with-package tool-mkspiffs

Can you check this in rfquack?

joelsernamoreno, 2021-04-03:
Can you try to install rfquack on any dongle using Docker?

I would like to know if it is a problem with rfquack, platformio or my computer. We have tried several colleagues and we all get the same error

Thanks

joelsernamoreno, 2021-04-04:
H!

This works for me:

1.- Create a Dockerfile on your computer with the following content:
[image]

2.- Compile this with: docker build -t evilcrowrf/rfquack .

3.- Flash firmware: docker run --device=/dev/ttyUSB0:/board -e BOARD=ESP32 -e RADIOA=CC1101 -e RADIOA_CS=27 -e RADIOA_IRQ=25 -e RADIOB=CC1101 -e RADIOB_CS=5 -e RADIOB_IRQ=2 --rm -it evilcrowrf/rfquack

To fix this for everyone you can update rfquack with this information. Platformio now does not allow to use older versions and therefore rfquack cannot install the necessary packages.

Thanks!

phretor, 2021-04-12:
@joelsernamoreno I'm looking into this, but I don't see `tool-mkspiffs` used anywhere in the default `Dockerfile`; if this is something specific to your flavor of `Dockerfile`, can you make a case why this should be addressed by RFQuack? If it's something blocking that doesn't conflict with RFQuack, I'm more than happy to help.

joelsernamoreno, 2021-04-12:
Hi!

When you run the command to flash rfquack with docker you are using espressif32. When you install espressif32 on platformio it automatically installs other packets like tool-mkspiffs

The problem is that your Dockerfile uses an old version of platformio and because of this it doesn't work correctly.

Change pip install -U platformio==4.1.0 to pip install -U platformio

phretor, 2021-04-12:
OK. Got it. That's what I'm about to commit. Thanks!

joelsernamoreno, 2021-04-12:
It seems that now platformio does not work properly if you are using older versions.

Let me know when you change this and I'll test it :D

phretor, 2021-04-12:
Have a look now @joelsernamoreno and let me know if it builds fine.

joelsernamoreno, 2021-04-12:
Hi!

I keep getting the same error

Have you changed this in the public container? Note that my installation is with docker run:

docker run --device=/dev/ttyUSB0:/board -e BOARD=ESP32 -e RADIOA=CC1101 -e RADIOA_CS=27 -e RADIOA_IRQ=25 -e RADIOB=CC1101 -e RADIOB_CS=5 -e RADIOB_IRQ=2 --rm -it rfquack/rfquack

I am not flashing this locally

phretor, 2021-04-12:
No. I only changed the `Dockerfile`. Just run `make docker-build` to have a fresh image built locally.

joelsernamoreno, 2021-04-12:
Well, this has compiled

How do I flash this now? I have installed the following:

docker run --device=/dev/ttyUSB0:/board -e BOARD=ESP32 -e RADIOA=CC1101 -e RADIOA_CS=27 -e RADIOA_IRQ=25 -e RADIOB=CC1101 -e RADIOB_CS=5 -e RADIOB_IRQ=2 --rm -it rfquack/rfquack:latest

Now I don't get errors from tool.mkspiffs, but I get these errors:

[image]

dgeppert, 2021-04-12:
these compilation errors look familiar to me, that's why I was using @FiorixF1 git repos.
it seems to me that there is some misalignment between RadioLib and RFQuack, 
but if @phretor manages to compile the error should be on my side, somewhere...

strange, using the latest Docker hub RFQuack image, I still get:

```
[dirk@t520 RFQuack]$ docker run --device /dev/ttyUSB0:/board -e BOARD=ESP32    -e RADIOA=CC1101 -e RADIOA_CS=5  -e RADIOA_IRQ=4  -l DEBUG --rm -it rfquack/rfquack:latest
Installation ID is 9b58cd12-5225-4f74-8097-14f3d1f0999f
Processing ESP32 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing tool-mkspiffs @ ~2.230.0
Error: Detected unknown package 'tool-mkspiffs'
```

joelsernamoreno, 2021-04-12:
Lol, now my previous solution doesn't work either

Seems to have broken everything :(

joelsernamoreno, 2021-04-12:
Hi @dgeppert 

With the solution I had written before (create a new Dockerfile with the information I attached) the tool-mkspiffs problem was solved

But... after today's new changes it seems to have broken everything.

dgeppert, 2021-04-12:
in my Dockerfile I successfully use:

```
RUN apt-get update \
  && apt-get install -y --no-install-recommends git wget unzip make \
  && apt-get purge -y --auto-remove \
  && rm -rf /var/lib/apt/lists/* \
  && pip install -U platformio \
  && platformio platform install espressif32 --with-package tool-mkspiffs \
  && pip install j2cli \
  && pip install google==2.0.3 \
  && pip install protobuf==3.11.3
```
hth

phretor, 2021-04-12:
@FiorixF1 could you please send PRs from your RadioLib fork to rfquack/RadioLib so I can realign?

joelsernamoreno, 2021-04-12:
@dgeppert 
Using your dockerfile I keep getting the RadioLib error

dgeppert, 2021-04-12:
hi @joelsernamoreno 
with the changes in Dockerfile only the mkspiffs issue was addressed according to your previous proposal

compilation issues will be solved once @phretor incorporates @FiorixF1 RadioLib changes and aligns RFQuack, I guess...

joelsernamoreno, 2021-04-12:
Oh, good. This is a radiolib problem

Sorry for trying to update this. A few weeks ago I started getting the tool-mkspiff error.

I solved this by creating an updated Dockerfile to use the latest version of platformio and this worked. 

I didn't think this problem would become so big now hahaha

dgeppert, 2021-04-12:
last comment: after rm all my docker images/containers a `docker run rfquack/rfquack:latest` does not show the mkspiffs error message. seems it did not pull the new image and reused the local one ;-( 

just the RadioLib compilation errors remain as @joelsernamoreno already reported

phretor, 2021-04-13:
@dgeppert @joelsernamoreno see the latest commits (and, in particular, the `Makefile` and newer `Dockerfile`) and additions to the `README.md` file and let me know if you're still having issues.

In short, you should be able to:

```bash
$ edit build.env
$ make docker-build
$ make build
$ PORT=/dev/ttyUSB0 make flash
```

joelsernamoreno, 2021-04-13:
Hello!

In the **make build** command I get the following errors:

[image]

phretor, 2021-04-13:
Check now please:

```bash
$ make docker-build-nc && make build

...

Compiling .pio/build/HUZZAH32/FrameworkArduino/esp32-hal-timer.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/esp32-hal-touch.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/esp32-hal-uart.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/main.cpp.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/HUZZAH32/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/HUZZAH32/libFrameworkArduino.a
Linking .pio/build/HUZZAH32/firmware.elf
Retrieving maximum program size .pio/build/HUZZAH32/firmware.elf
Checking size .pio/build/HUZZAH32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  16.9% (used 55344 bytes from 327680 bytes)
Flash: [==        ]  22.9% (used 299594 bytes from 1310720 bytes)
Building .pio/build/HUZZAH32/firmware.bin
esptool.py v3.0
=============================== [SUCCESS] Took 33.12 seconds ===============================
```

joelsernamoreno, 2021-04-13:
Hi!

It compiles now, but I can't connect to the dongle:

[image]

This happens when running the command docker run --device /dev/ttyUSB0 --rm -it rfquack/cli tty -P /dev/ttyUSB0

I get the error ModuleNotFoundError: No module named 'click' but this is installed on my machine

joelsernamoreno, 2021-04-13:
I have tried to install RFQuack-cli locally, but there are ipython errors:

1. git clone https://github.com/rfquack/RFQuack-cli.git
2. cd RFQuack-cli/
3. pipenv install -e .

[image]

phretor, 2021-04-13:
There was indeed a missing line in the `Dockerfile`: thanks for your patience.

When installing locally, beware that you're using Python < 3.9 (which is the recommended minimum version). Because of that, IPython could not be installed.

I recommend using Pyenv to manage multiple Python versions on the same system, if you really want to go dockerless for this one.

phretor, 2021-04-13:
Closing this issue. If the CLI thing reappears, please open an issue at rfquack/RFQuack-cli.

closed 2021-04-13 by phretor

joelsernamoreno, 2021-04-13:
Hi!

This work:
1. cd RFQuack-cli
2. make docker-build
3. docker run --device /dev/ttyUSB0 --rm -it rfquack/cli:latest tty -P /dev/ttyUSB0

Thanks!