rfquack/RFQuack #7: capture cli mesages

dgeppert, 2021-04-12:
Hello,

not an issue, but a question (hope that's ok here):

is there a way to capture the cli result into e.g. an array to process the result later within the cli?

```
RFQuack(/dev/ttyUSB0)> a = q.radioA.get_register(0x71)                                                                                               

address = 113
value = 4
0x71 = 0b00000100 (0x04, 4)

result = 0
message = 

RFQuack(/dev/ttyUSB0)> a                                                                                                                         

RFQuack(/dev/ttyUSB0)> print(a)                                                                                                                  
None
```
redirecting stdout does not seem to help here, maybe because Protobuf is used?

thx in advance

phretor, 2021-04-12:
@dgeppert results are automatically collected into the `q.data` variable, which is of type `list()`.

dgeppert, 2021-04-12:
```
RFQuack(/dev/ttyUSB0)> q.radioA.get_register(0x71)                                                                                               

address = 113
value = 4
0x71 = 0b00000100 (0x04, 4)

result = 0
message = 

RFQuack(/dev/ttyUSB0)> q.data                                                                                                                    
Out[2]: []

RFQuack(/dev/ttyUSB0)> print(q.data)                                                                                                             
[]
```

phretor, 2021-04-12:
Oh, my bad. You want to collect the register values. I don't think we've ever thought about storing those.

Check here https://github.com/rfquack/RFQuack-cli/blob/master/rfquack/core.py#L237

You'll see that we're storing only packets (e.g., transmissions) into `self.data`. Maybe you can create a `self.register = list()` and collect `Register` objects at https://github.com/rfquack/RFQuack-cli/blob/master/rfquack/core.py#L231

If you can do that and send a PR, I'll merge it ;-)

Guglio95, 2021-04-12:
Hello @dgeppert ,
‘q.data’ stores packets received from the transceiver while in RX mode

dgeppert, 2021-04-12:
thanks @phretor! for pointing to the right spots ;-)

just two lines of code in rfquack/core.py

```
class RFQuack(object):
        ...
        self.data = list()
        self.register = list() # <---

        ...

        if isinstance(msg, rfquack_pb2.Register):
            fmt = '0x{addr:02X} = 0b{value:08b} (0x{value:02X}, {value})\n'
            out += fmt.format(
                **dict(addr=msg.address, value=msg.value))
            self.register.append(msg)) # <---
```

```
RFQuack(/dev/ttyUSB0)> q.radioA.get_register(0x71)                                                                                               

address = 113
value = 4
0x71 = 0b00000100 (0x04, 4)

result = 0
message = 

RFQuack(/dev/ttyUSB0)> q.register                                                                                                                
Out[2]: 
[address: 113
 value: 4]
```

closed 2021-04-12 by phretor