rfquack/RFQuack #9: RFQuack does not work properly

joelsernamoreno, 2021-04-29:
Hi!

I can't access the RFQuack console correctly after flashing the code. My steps are as follows:

1.- git clone https://github.com/rfquack/RFQuack.git
2.- cd RFQuack
3.- I edit build.env with the following content:
* BOARD=ESP32
* RADIOA=CC1101
* RADIOA_CS=27
* RADIOA_IRQ=25
* RADIOB=CC1101
* RADIOB_CS=5
* RADIOB_IRQ=2

4.-  make docker-build-nc && make build
5.- PORT=/dev/ttyUSB0 make flash

RFquack is successfully flashed on the device

6.- docker run --device /dev/ttyUSB0 --rm -it rfquack/cli tty -P /dev/ttyUSB0

Device not detected correctly:

[image]

Thanks!

phretor, 2021-05-03:
> The last commit that works fine is this one: git checkout 9824555

Did you check this? https://github.com/rfquack/RFQuack-cli/issues/7#issuecomment-830572520

> Additionally I have noticed that RFQuack has poor reception, I have to be very close to RFQuack to receive any remote control. Where do you configure the CC1101 registers? Possibly by changing this you can receive signals at a greater distance.

There are many settings that can affect reception quality, but playing with `txPower` (from the CLI) is what I recommend doing first. Via RFQuack CLI you can access every single register:

```python
q.radioA.register(address=0x..., value=0x...)
```

which, on the other side, calls these `set_register` and `get_register` methods: https://github.com/rfquack/RFQuack/blob/master/src/modules/defaults/RadioModule.h#L43

phretor, 2022-08-27:
Closing for no activity.

closed 2022-08-27 by phretor