rfquack/RFQuack #15: Building with "make docker-build" and "make docker-build-nc" fails

canidorichard, 2022-01-23:
Attempting to build with "make docker-build" or "make docker-build-nc" fails with the following error:

make: *** No rule to make target 'proto'.  Stop.
The command '/bin/sh -c platformio lib -g install file:///tmp/RFQuack   && cd $HOME/.platformio/lib/RFQuack/   && make proto' returned a non-zero code: 2
make: *** [Makefile:47: docker-build-nc] Error 2

canidorichard, 2022-01-24:
I can see from an earlier commit that the proto target was removed from the Makefile and so I removed the make proto from Docker file and the creation of the image is successful however it looks like there is some clean up required around protbuf as the next command ("make build") fails as a result of missing header file:
```

  /root/.platformio/lib/RFQuack/src/radio/../modules/../rfquack_common.h:36:24: fatal error: rfquack.pb.h: No such file or directory

********************************************************************
* Looking for rfquack.pb.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:rfquack.pb.h"
* Web  > https://platformio.org/lib/search?query=header:rfquack.pb.h
*
********************************************************************

compilation terminated.
```

phretor, 2022-01-26:
@canidorichard thanks a lot for checking RFQuack. As an experimental project, it takes folks like you to play around and file issues :-)

The `make proto` is needed to build `rfquack.pb.h` but it fails because (I think) I forgot to include the installation of protobuf.

It should be there https://github.com/rfquack/RFQuack/blob/feee974f147f0797ad0b0d275f1cd29b2a1a55db/Dockerfile but you may want to double check.

canidorichard, 2022-01-26:
HI @phretor I am very happy to help where I can as this is an interesting project.  I suspected it needed protobuf but noticed you removed "make proto" from the Makefile in your commit of 24th September 2021 and so that confused me a bit.

phretor, 2022-01-26:
@canidorichard let me know how it goes. I'm happy to give directions and help wherever I can. I just have the time to properly "sit down" and work on this project, although I think it has great potential, especially as an educational tool to teach the basics of digital RF comm. It's been used at a local school, although they struggle (like many schools at the moment) and didn't have the time to produce any public-facing material.

canidorichard, 2022-01-26:
I have restored the proto target in Makefile which has resolved all of my build issues although I have not flashed the firmware and tested it as yet.  I am happy to create a pull request but wanted to understand why it had been removed previously on 24th September 2021 as I feel I may be missing some reasoning in which case me undoing this change may not be entirely helpful.

phretor, 2022-01-26:
It was probably due to a glitch in my refactoring. I usually work in bursts on side projects and I tend to leave some things behind.

My idea was to start paving the way to some GitHub Actions to do automated compilation/checks, but I never completed that step.

Feel free to send any PRs, of course :-)

ResistanceIsUseless, 2022-08-20:
I'm running into this same issue with `No rule to make target 'proto'` I'm not really familiar with what's going on here. What rule is it looking for and what generates it? If you can give me some guidance I'll submit a PR.

I also ran into some issues with some libraries needing updating the the platformio command changing. I'll submit a PR for those updates.

ResistanceIsUseless, 2022-08-21:
I think the issue I'm running into is most of the proto files are missing. I've been digging through some of the manual steps you provided in other closed issues and if i can get around the `make proto` issue and get `rfquack.pb.h` generated i should be ok.

Once this is all figured out it would be nice to have actions generate the firmware for future builds. I think that would get more people interested in testing out this framework and getting more contributors. Right now it's a lot to swallow and has a high bar for entry when the containers build fails.

phretor, 2022-08-21:
Thanks for contributing with your comments and ideas.

I know I sound like a broken record, but the time I can dedicate to this project is almost  zero. But I really want to get this project on the right track, and investing time on creating Actions is a good way.

ResistanceIsUseless, 2022-08-21:
All good, that's open source life. I finally got the docker builds to work, honestly not even sure what did it. Possibly updating bottle or a different machine? I'll submit a PR to at least get the pip versions i updated fixed.

I'm now stuck trying to get firmware to flash after building. I'll create a new issue for that if it's something i get stuck on.

phretor, 2022-11-16:
Closing. New build system.

closed 2022-11-16 by phretor