rfquack/RFQuack #19: Choose document generator system

phretor, 2022-08-27:
I'm asking to RFQuack's friends to help me choose the doc-generation system. Here are the options I know:

- GitBook
  - PROs: easy to get started with, Markdown based
  - CONs: kinda proprietary

- Docusaurus:
  - PROs: fully open source, Markdown based, simple
  - CONs: none known to me

- MkDocs
  - PROs: Markdown based, active community, many themes
  - CONs: maybe old?

- Sphinx:
  - PROs: well supported, mature product
  - CONs: old, Python centric

ResistanceIsUseless, 2022-08-27:
I've personally used Hugo and found it pretty straight forward. It's based on Go and uses Go Templating.

ResistanceIsUseless, 2022-08-27:
Also heard https://astro.build/ is nice.

phretor, 2022-08-28:
Hugo and Astro are static site generators. I was looking for doc-specific generators.

phretor, 2022-11-24:
MkDocs is the chosen one.

closed 2022-11-24 by phretor