rfquack/RFQuack #20: Look into using ESPHome as method for generating / uploading / managing firmware

ResistanceIsUseless, 2022-08-27:
I know RFQuack isn't really ESP specific since it's supports anything RadioLib supports. I'm thinking that using ESP32's as the primary method makes the most sense though. I don't know if you have used ESPHome, but it's seriously easy and so straight forward.

This is what I'm thinking as to why:
- ESPHome supports over the air and in browser flashing.
- ESPHome uses a yaml based templating language that would simplify setup.

Essentially if we could make it so ESPHome load RFQuack as a library and handle the configuration of board, radio(s) through the yaml it would be much easier.

phretor, 2022-08-28:
Yes, having ESPHome as one of the build systems for RFQuack-based dongles can be useful, although it wasn't the main idea of this project. I'd keep this idea in the bucket list but would prioritize.

phretor, 2023-02-02:
Closing for no activity.

closed 2023-02-02 by phretor