rfquack/RFQuack #30: getSyncWord - invalid conversion

gigawatts, 2022-11-28:
## Describe the bug
Trying to compile RF Quack for EvilCrowRF v1

Getting the below error
```
src/radio/RFQCC1101.h: In member function 'virtual int16_t RFQCC1101::getSyncWord(uint8_t*, pb_size_t*)':
src/radio/RFQCC1101.h:103:16: error: invalid conversion from 'int' to 'pb_size_t*' {aka 'short unsigned int*'} [-fpermissive]
         size = 2;
                ^
```
Which is similar to #2 but that was for `setSyncWord` function, this is for `getSyncWord` function, and the workaround listed in that issue did not work for this problem.

## To Reproduce

Using a combo of instructions from the following locations:

- https://github.com/joelsernamoreno/EvilCrow-RF#advanced-firmware-with-rfquack (Tried the docker build method there, although the make options differ from the  one listed in that readme)
- http://rfquack.org/usage/dependencies/
- http://rfquack.org/usage/dependencies/ (I tried both the poetry and non-poetry install methods)

## Expected behavior

Expect firmware to build and flash

## Additional info (please complete)

- MCU: ESP32 (EvilCrowRF v1)
- Wireless module type:  CC1101
- Host environment:  Fedora 36 64bit
- RFQuack version, branch, or tag:  master
- RadioLib version, branch, or tag:  master

gigawatts, 2022-11-28:
strange, cant seem to edit above text beyond "To Reproduce"

RFQuack commit hash: `aa7f537d587d18e6795406e26f02947b6da6c089`
RadioLib commit hash: `b9a7116113663b69c6d6f7e69b925d3cdfd860cf`

gigawatts, 2022-11-28:
Also note that the instructions at the following link don't work, as the freshly built docker image does not have `pio` installed
http://rfquack.org/usage/build/#build-via-docker 

```
Successfully built 29f9cca96499
Successfully tagged rfquack/rfquack:latest

$ make build-in-docker
docker run \
	--rm -it \
	--volume /home/me/Downloads/RFQuack:/tmp/RFQuack rfquack/rfquack \
	make build
pio run
/bin/bash: line 1: pio: command not found
make: *** [Makefile:82: build] Error 127
make: *** [Makefile:69: build-in-docker] Error 2
```

phretor, 2022-11-28:
Thanks @gigawatts - please have a look at the most recent commit.

closed 2022-11-28 by phretor

gigawatts, 2022-11-29:
Working! Thank you!

commit 0a315a90 (2022-11-28): Fix #30