rfquack/RFQuack #32: ECRFv2: Building bug with ECRF set

AkaBkn, 2022-12-01:
## Describe the bug

**lib/RadioLib/src/Module.h:49**
```Module(RADIOLIB_PIN_TYPE cs, RADIOLIB_PIN_TYPE irq, RADIOLIB_PIN_TYPE rst, RADIOLIB_PIN_TYPE gpio, SPIClass& spi, SPISettings spiSettings = RADIOLIB_DEFAULT_SPI_SETTINGS);```

**docker/project/src/main.cpp.j2**
```
RadioA radioA = new Module(
  {{ RADIOA_CS }},
  {{ RADIOA_IRQ|default('RADIOLIB_NC') }},
  {{ RADIOA_RST|default('RADIOLIB_NC') }}

{% if ECRF is defined %}
  , SPIH, spiSettings
{% endif %}
```

main.cpp.j2 misses the `gpio` argument

## To Reproduce

having build.env:
```
BOARD=ESP32
ECRF=true

#RADIOA=CC1101
#RADIOA_CS=27
#RADIOA_IRQ=25

RADIOA=CC1101
RADIOA_CS=5
RADIOA_IRQ=2
RADIOA_GP=4

RADIOB=CC1101
RADIOB_CS=27
RADIOB_IRQ=25
RADIOB_GP=26

RADIOC=nRF24
RADIOC_CS=15
RADIOC_CE=33
RADIOC_IRQ=7
```

`make clean build`

## Expected behavior

compiles fine

phretor, 2022-12-02:
Please check the documentation for the allowed variables. _CE or _GP don't mean much to RFQuack.

phretor, 2022-12-02:
@AkaBkn please check - it compiles on my side.

AkaBkn, 2022-12-02:
@phretor looks good, thx :) btw. would be cool to have an example `build.env` for ECRFv2 with nRF24 too

phretor, 2022-12-02:
I'll test it when I'll receive my ECRFv2

joelsernamoreno, 2022-12-11:
Mm... Error: Unknown development platform 'espressif32'??

phretor, 2022-12-11:
> Mm... Error: Unknown development platform 'espressif32'??

Have you tried to change the platform value into platformio.ini ?

joelsernamoreno, 2022-12-11:
Fix: platformio upgrade --dev

gisforgirard, 2023-02-04:
not sure why you have RADIOC_.... set, perhaps see https://github.com/joelsernamoreno/EvilCrow-RF for more detailed instructions:

[image]

temskiy, 2023-06-19:
I add [PR](https://github.com/rfquack/RFQuack/pull/47) with tested build.env for ECRFv2

phretor, 2023-06-19:
Awesome. Thanks @temskiy!

closed 2023-06-19 by phretor