rfquack/RFQuack #34: Pinout?

Crsarmv7l, 2023-01-23:
Hi,

Looking to get started with rfquack. I have an ESP32 and cc1101 on hand and just soldered the headers to the esp32. What I can't seem to find after looking through all the docs and the wiki is the pinout.

I have my cc1101 pinout
[image]

and I have my esp32 pinout (on the board). What needs to be connected? I understand assigning the pins in the build env.

phretor, 2023-01-23:
@Crsarmv7l good point. You need to have the radio (CC1101) correctly connected to the SPI bus of the MCU (ESP32). So, at the bare minimum, you need this:

- Power
  - Vcc
  - GND

- SPI common wires
  - MOSI to the MCU pin marked as "MOSI"
  - MISO to the MCU pin marked as "MISO"
  - CLK  to the MCU pin marked as "CLK"

- SPI channel select
  - CSn connected to a clear MCU pin (i.e., not used by anything else)

- Interrupts (at least one)
  - GDO0 to an interruptible MCU pin
  - GDO2 to another interruptible MCU pin

You'll find pinouts (which are specific to your board) by googling for (example) "Feather HUZZAH32 pinout" ([relevant result](https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/pinouts)).

Feel free to contribute back into the documentation with what you think may be useful to other users!

Crsarmv7l, 2023-01-23:
Thank you for the response. Would you mind leaving this open for a a few days while I try getting it working? I will close it once successful.

Got my detailed pinout will give it a shot in the next few days
[image]

Crsarmv7l, 2023-01-24:
While I am working on this I do have a question regarding the build env. In certain lines there are multiple definitions of RadioA_RST, they are * in the below code. I assume this is an oversight and they should be for RadioB/etc?

```
# These settings are not guaranteed to work: they're only for CI/CD purposes

RADIOA=RF69
RADIOA_CS=13
RADIOA_IRQ=26
*RADIOA_RST=32*

RADIOB=RF69
RADIOB_CS=33
RADIOB_IRQ=25
*RADIOA_RST=27*

RADIOB=nRF24
RADIOB_CS=34
RADIOB_IRQ=26
*RADIOA_RST=28*
```

Crsarmv7l, 2023-01-24:
@phretor 

Ah yep looks like GDO = IRQ

The question though does CSn connect to CS or an unused pin? My ESP32 has a CS/SS pin.

phretor, 2023-01-24:
Yes, you can use the CS pin if your board comes with one.

Yes, that was an oversight.

Crsarmv7l, 2023-01-29:
I believe I have my pinout correct, and I have flashed my board, however the rfq command is not found. Some further digging and it appears to be in the docker image.

I think at this point I am going to step back from attempting to get this going. I will circle back in a few months.

closed 2023-01-29 by Crsarmv7l