rfquack/RFQuack #35: Unable to set modulation on EVRFv2

bazuchan, 2023-01-24:
I'm using RFQuack on EvilCrowRFv2 with mqtt transport. I can set some options for CC1101 radio (freq, rate, bandwith) just fine, but I can't set modulation.

```
RFQuack(RFQuackShell, 172.17.2.1:1883)> q.radioA.set_modem_config(modulation="OOK")

result = -1
message = 0 changes applied and 1 failed.
```

Debug log on device
```
[RFQK]   58500337 T: Got command for moduleName: radioA, verb: set, argsLen: 2, messageLen 2
[RFQK]   58500338 E: Unable to set modulation
[RFQK]   58500339 T: Transport is sending 44 bytes on topic rfq1/out/set/radioA/rfquack_CmdReply/set_modem_config
```

My hardware config is https://github.com/joelsernamoreno/EvilCrowRF-V2/issues/18

phretor, 2023-01-24:
@bazuchan can you please try to enable (really) verbose debug output in RadioLib here? https://github.com/jgromes/RadioLib/blob/master/src/BuildOpt.h#L924

Of course you should do that before running any build command and ensuring that a fresh copy of RadioLib is **not** getting fetched, or it'd overwrite your changes.

bazuchan, 2023-01-24:
Looks like no data is sent to radio when trying to set modulation.

```
RFQuack(RFQuackShell, 172.17.2.1:1883)> q.radioA.set_modem_config(bitRate=10)

result = 0
message = 1 changes applied and 0 failed.

RFQuack(RFQuackShell, 172.17.2.1:1883)> q.radioA.set_modem_config(modulation="OOK")

result = -1
message = 0 changes applied and 1 failed.
```

```
baz@butterfly:~/RFQuack$ make console | grep -vP '^R\t7\tE'
pio device monitor
--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H

[RFQK]      63561 T: Got command for moduleName: radioA, verb: set, argsLen: 2, messageLen 5
R       10      A8
W       10      A8
R       10      A8
R       11      93
W       11      93
R       11      93
[RFQK]      63573 T: Transport is sending 35 bytes on topic rfq1/out/set/radioA/rfquack_CmdReply/set_modem_config
[RFQK]      65918 T: Got command for moduleName: radioA, verb: set, argsLen: 2, messageLen 2
[RFQK]      65919 E: Unable to set modulation
[RFQK]      65920 T: Transport is sending 44 bytes on topic rfq1/out/set/radioA/rfquack_CmdReply/set_modem_config
```

phretor, 2023-01-24:
Likely, something is going sideways here:

https://github.com/rfquack/RFQuack/blob/3a493da01f9c0b0bab76469fdc0f1d4385863b9f/src/radio/RFQCC1101.h#L168

Maybe you can play around and add some extra debug messages to see what branch actually fails.

bazuchan, 2023-01-24:
Found it. I'm using https://github.com/rfquack/RFQuack-cli and it looks like deprecated and has other values in protobuf.
But I got it from installation instructions here https://github.com/rfquack/RFQuack/tree/master/client

phretor, 2023-01-24:
Thanks: https://github.com/rfquack/RFQuack/commit/943c1f231b2ff39269759a45efb6bb7121b70025

closed 2023-01-24 by bazuchan