rfquack/RFQuack #39: Can RFQuack be used with RF1100-232 USB device ?

arthurlutz, 2023-03-13:
## Describe the bug

I am trying to use RFQuack with an RF110-232 USB device such as https://aliexpress.com/item/1005003285561543.html I can see it as /dev/ttyUSB0 so typing this : 

```
PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python rfquack tty -P /dev/ttyUSB0 -b 19200 
# or with futher options
PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python rfquack tty -P /dev/ttyUSB0 -b 19200 -p N -S 1 -s 8
```

RFQuack start correctly but no dongles are detected (`q._dongles` returns a `{}` so I can't select by id). 

I have access to some software to configure the CC1101 chip via the USB / CP2102 USB to UART Bridge Controller (Silicon Labs). This device seems to be build by a defunct company by the name of fytoo. Firmware and PDFs (in chinese) are to be found here https://github.com/acontini/CC1101_USB

Am I trying to do something completely out of the scope of RFQuack ? 

## Additional info (please complete)

- MCU: ttyUSB via CP2102 (?)
- Wireless module type : CC1101
- Host environment : Linux Ubuntu 22.02
- RFQuack version, branch, or tag :  master
- RadioLib version, branch, or tag : n/a ?

phretor, 2023-03-14:
Not out of the box. The radio transceiver CC1101 is supported by RFQuack, but the board may not be supported by PlatformIO and/or Arduino.

Do you know what MCU is onboard the RF110-232?

Have you checked it it's supported by PlatformIO?

arthurlutz, 2023-03-15:
The MCU seems to be an ATMEL MEGA48PA. It seems supported by plateformio : https://docs.platformio.org/en/latest/boards/atmelavr/ATmega48P.html

phretor, 2023-03-15:
OK, that's half good news. Now, in order to use RFQuack you need to figure
out how the MCU's SPI bus is wired to the CC1101.

arthurlutz, 2023-03-15:
Here are a few quick picture, if this a way forward I can probably do some higher resolution pictures to make it easier to read. 

[image]
[image]
[image]
[image]
[image]
[image]

I believe some information about this USB stick is archived here : https://github.com/acontini/CC1101_USB/ (most documents in Chinese) 

Another source of information is https://github.com/acontini/CC1101_USB/

phretor, 2023-03-15:
I think you don’t need to follow the wiring. Instead, you can figure out
the pinout from that source code.

jradm87, 2023-05-04:
Hello, @arthurlutz , let me know if you got it running, I have the same device.

adminy, 2024-05-25:
me too! Let us know.