rfquack/RFQuack #40: Please help with my pinout

Crsarmv7l, 2023-03-24:
Finally got back to trying to get rfquack working, and after compiling and installing python 3.10.10 on my debian machine, working through the oddities of the build (not using poetry to install depends, only pip with the requirements), etc, my board is flashed and the interactive environment works.
..
   However, I can only access q.dongle... Which means I likely wired something wrong with my cc1101. Its late and this has taken me a while so I can't say I am surprised. I was hoping for another set of eyes to take a look please.

I am using an ebyte cc1101 with this pinout:

[image]

and I am using a WeMoS Lolin32 with this pinout:

[image]

I connected vcc to 3v, ground to ground, and MISO/MOSI/SCK/and CS to these:

[image]

In build.env RadioA_CS is 34

[image]

RadioA_IRQ (GD0) is 27

[image]

Really just hoping I made a silly mistake with the wiring, and it will be a simple rewire, reflash and I can finally give this a go.

Thanks

phretor, 2023-03-24:
Wiring and configuration look in order. Can you share a dump of the console output at boot, possibly with log/dev/debug enabled?

Define the relevant variables in your build.env - the complete set of build variables can be guessed from here - https://github.com/rfquack/RFQuack/blob/master/docker/project/src/main.cpp.j2

Crsarmv7l, 2023-03-24:
Well that solved it. Unfortunately. pio device monitor showed it was constantly browning out. It exhibited that behavior even with gpio/cc1101 completely disconnected.

I do have another board, a feather ESP32-S3 with no PSRAM and I have tried building RFquack for that ( with adafruit_feather_esp32s3_nopsram  for board in platformio.ini) but it fails every time with the following:

[build.txt](https://github.com/rfquack/RFQuack/files/11062551/build.txt)

Crsarmv7l, 2023-04-04:
Picked up an original feather esp32. So I'll close this and hopefully finally be able to get this to work

closed 2023-04-04 by Crsarmv7l