rfquack/RFQuack #42: Python example

Crsarmv7l, 2023-04-04:
I am aware this is scriptable using python and very similar to the Yardstick 1. Can you provide an example script just to make sure I am working with it correctly

phretor, 2023-04-05:
This is the most complete and working example you'll find: https://github.com/rfquack/RFQuack/tree/master/examples/demos

If you're interested in the recorded demo so you can follow along: https://youtu.be/XGDuEhhK6gQ

Crsarmv7l, 2023-04-05:
> This is the most complete and working example you'll find: https://github.com/rfquack/RFQuack/tree/master/examples/demos
> 
> If you're interested in the recorded demo so you can follow along: https://youtu.be/XGDuEhhK6gQ

Thank you I will check it out. Being able to write python scripts for the YS1 is why I liked it so much. With the poetry/python env I wasn't sure if that would affect the writing of a python script for rfquack.

Hopefully I can get a better understanding from the demo

phretor, 2023-04-05:
It's just Python. Pyenv and Poetry are just modern Python dev practice.

closed 2023-04-07 by Crsarmv7l