rfquack/RFQuack #44: Tx works, Rx doesn't

Crsarmv7l, 2023-04-09:
## Describe the bug

Like others on discord I can get rfquack to transmit using the feather esp32 and CC1101, but I get nothing, not even noise when receiving.
## To Reproduce

Using a pull from the latest

## Expected behavior

Both Rx and Tx work

## Additional info (please complete)

- MCU: Feather ESP32
- Wireless module type: CC1101
- Host environment: Linux, Debian 11, Python 3.10.10 
- RFQuack version, branch, or tag: Master

Crsarmv7l, 2023-04-10:
On compile I get this warning:

`In file included from src/radio/RFQCC1101.h:4,
                 from src/radio/drivers.h:4,
                 from src/main.cpp:67:
src/radio/RadioLibWrapper.h: In member function 'int16_t RadioLibWrapper<T>::receiveMode() [with T = CC1101]':
src/radio/RadioLibWrapper.h:137:3: warning: control reaches end of non-void function [-Wreturn-type]
   }
   ^
`

Crsarmv7l, 2023-04-11:
I verified that the rfquack package is able to transmit with an SDR.

I will have to look at console later today to see if there are any messages with receive.

phretor, 2023-04-11:
You may want to set it in promiscuous mode.

Crsarmv7l, 2023-04-12:
> You may want to set it in promiscuous mode.

According to this issue https://github.com/rfquack/RFQuack/issues/36 promiscuous is broken on the cc1101

Crsarmv7l, 2023-04-12:
> Can you validate with console output and (better) with an SDR?

Sorry, I thought I would be able to log using pio device monitor in one window while issuing commands over the other. I am getting an error that something is already connected when using pio device monitor and trying to issue commands (I am using tty to issue commands).

Am specific way I can log while issuing commands over tty?

phretor, 2023-04-13:
You should check what else is using that serial port

Crsarmv7l, 2023-04-13:
> You should check what else is using that serial port

Yes, I am connecting to the dongle via wire/serial (rfq tty -P /dev/ttyUSB0). 

Serial protocol only allows one connection. Pio device monitor would be a second connection.

Crsarmv7l, 2023-06-03:
Closing for now. I haven't done much with it. Been focused on the YS1. Will circle back again at some point

bazuchan, 2023-06-05:
Why closing it? I'm affected by same bug.

fabrein, 2023-06-05:
Same here. Send is working but RX does not receive anything.

Crsarmv7l, 2023-06-05:
Reopening. I made further attempts without success. I am fairly sure the Promiscuous issue is also at least part of the cause here.

Crsarmv7l, 2024-05-06:
Circling back. I have been deep in both Radiolib, CC1101 Datasheet, Ti Forums and other sources for a while now.

Couple things:
  - The FIFO's (which this uses) don't work with Syncword/preamble disabled
     - Sync/Async Serial has to be used with GDO0 and GDO2 if you want the actual raw stream. See Ti Developer Note AN095 Continuous Data Streaming
  - PQT was not set to Zero in Radiolib Promiscuous, I put in a pull to correct that
  - I have submitted a few other pulls to improve stuff in Radiolib

closed 2024-05-06 by Crsarmv7l

jstockdale, 2025-02-06:
Has anyone successfully confirmed rx works? It's still not working for me as far as I can tell. Thanks!