rfquack/RFQuack #45: error while building RFquack for ESP32C3

woahotter, 2023-04-27:
## Describe the bug

while attempting to build (with every dependency installed), it fails with:
```
... lots of compiling, no errors or warnings ...
Compiling .pio/build/seeed_xiao_esp32c3/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/seeed_xiao_esp32c3/FrameworkArduino/main.cpp.o
Compiling .pio/build/seeed_xiao_esp32c3/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/seeed_xiao_esp32c3/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/seeed_xiao_esp32c3/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/seeed_xiao_esp32c3/libFrameworkArduino.a
Indexing .pio/build/seeed_xiao_esp32c3/libFrameworkArduino.a
Linking .pio/build/seeed_xiao_esp32c3/firmware.elf
/home/otter/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/seeed_xiao_esp32c3/libFrameworkArduino.a(main.cpp.o): in function `loopTask(void*)':
/home/otter/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/seeed_xiao_esp32c3/libFrameworkArduino.a(main.cpp.o): in function `loopTask(void*)':
/home/otterworks-vm/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:47: undefined reference to `setup()'
/home/otter/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /home/otter/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:47: undefined reference to `loop()'

```

## To Reproduce

i just tried attempting building, and it doesn't work - installed all prerequisites from the rfquack site

## Expected behavior

successful building of rfquack, then flashing

## Screenshots or pictures

none

## Additional info (please complete)

- MCU: seeed xiao esp32c3
- Wireless module type: cc1101
- Host environment: fedora
- RFQuack version, branch, or tag: master
- RadioLib version, branch, or tag [e.g., v.5.3.0]

phretor, 2023-04-27:
Can you checkout a fresh copy of the repository somewhere else and type here the exact sequence of commands that you type, step by step?

Please paste the entire output, even when not containing errors.

woahotter, 2023-04-27:
following the instructions starting at https://rfquack.org/usage/download/ :
```
[root@otter]~# git clone --recursive https://github.com/rfquack/RFQuack   
Cloning into 'RFQuack'...
remote: Enumerating objects: 2171, done.
remote: Counting objects: 100% (257/257), done.
remote: Compressing objects: 100% (66/66), done.
remote: Total 2171 (delta 187), reused 212 (delta 168), pack-reused 1914
Receiving objects: 100% (2171/2171), 41.16 MiB | 35.06 MiB/s, done.
Resolving deltas: 100% (1125/1125), done.
Submodule 'doxygen/doxygen-awesome' (https://github.com/jothepro/doxygen-awesome-css.git) registered for path 'doxygen/doxygen-awesome'
Submodule 'lib/RadioLib' (https://github.com/rfquack/RadioLib) registered for path 'lib/RadioLib'
Cloning into '/root/RFQuack/doxygen/doxygen-awesome'...
remote: Enumerating objects: 2155, done.        
remote: Counting objects: 100% (534/534), done.        
remote: Compressing objects: 100% (145/145), done.        
remote: Total 2155 (delta 388), reused 493 (delta 366), pack-reused 1621        
Receiving objects: 100% (2155/2155), 8.05 MiB | 15.53 MiB/s, done.
Resolving deltas: 100% (1582/1582), done.
Cloning into '/root/RFQuack/lib/RadioLib'...
remote: Enumerating objects: 14533, done.        
remote: Total 14533 (delta 0), reused 0 (delta 0), pack-reused 14533        
Receiving objects: 100% (14533/14533), 5.70 MiB | 11.40 MiB/s, done.
Resolving deltas: 100% (9977/9977), done.
Submodule path 'doxygen/doxygen-awesome': checked out 'a3c119b4797be2039761ec1fa0731f038e3026f6'
Submodule path 'lib/RadioLib': checked out 'aa8330cf571a40241cde30e9f4cb710244d27495'
[root@otter]~# sudo dnf install protobuf-compiler
Fedora 37 - x86_64 - Updates                                                                                   47 kB/s |  21 kB     00:00    
Fedora Modular 37 - x86_64 - Updates                                                                           68 kB/s |  24 kB     00:00    
Package protobuf-compiler-3.19.6-1.fc37.x86_64 is already installed.
Dependencies resolved.
Nothing to do.
Complete!
[root@otter]~# cd RFQuack 
[root@otter]~/RFQuack# pip install -U platformio
Collecting platformio
  Downloading platformio-6.1.6.tar.gz (236 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 236.4/236.4 kB 3.8 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting bottle==0.12.*
  Downloading bottle-0.12.25-py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.2/90.2 kB 18.5 MB/s eta 0:00:00
Requirement already satisfied: click<9,>=8.0.4 in /usr/lib/python3.11/site-packages (from platformio) (8.1.3)
Collecting colorama
  Downloading colorama-0.4.6-py2.py3-none-any.whl (25 kB)
Collecting marshmallow==3.*
  Downloading marshmallow-3.19.0-py3-none-any.whl (49 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 49.1/49.1 kB 10.6 MB/s eta 0:00:00
Collecting pyelftools<1,>=0.27
  Downloading pyelftools-0.29-py2.py3-none-any.whl (174 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 174.3/174.3 kB 21.4 MB/s eta 0:00:00
Collecting pyserial==3.5.*
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 15.8 MB/s eta 0:00:00
Requirement already satisfied: requests==2.* in /usr/lib/python3.11/site-packages (from platformio) (2.28.1)
Collecting semantic_version==2.10.*
  Downloading semantic_version-2.10.0-py2.py3-none-any.whl (15 kB)
Collecting tabulate==0.9.*
  Downloading tabulate-0.9.0-py3-none-any.whl (35 kB)
Collecting aiofiles==22.1.*
  Downloading aiofiles-22.1.0-py3-none-any.whl (14 kB)
Collecting ajsonrpc==1.*
  Downloading ajsonrpc-1.2.0-py3-none-any.whl (22 kB)
Collecting starlette==0.23.*
  Downloading starlette-0.23.1-py3-none-any.whl (64 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 64.5/64.5 kB 15.6 MB/s eta 0:00:00
Collecting uvicorn==0.20.*
  Downloading uvicorn-0.20.0-py3-none-any.whl (56 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 56.9/56.9 kB 10.3 MB/s eta 0:00:00
Collecting wsproto==1.2.*
[... 457 lines cut ...]
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.h:55:10: note: candidate: 'void HWCDC::begin(long unsigned int)'
     void begin(unsigned long baud=0);
          ^~~~~
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/HWCDC.h:55:10: note:   candidate expects 1 argument, 2 provided
Compiling .pio/build/seeed_xiao_esp32c3/libb08/ArduinoLog/ArduinoLog.cpp.o
Archiving .pio/build/seeed_xiao_esp32c3/lib9f4/libQueue.a
Indexing .pio/build/seeed_xiao_esp32c3/lib9f4/libQueue.a
In file included from src/radio/RFQCC1101.h:4,
                 from src/radio/drivers.h:4,
                 from src/main.cpp:53:
src/radio/RadioLibWrapper.h: In member function 'int16_t RadioLibWrapper<T>::receiveMode() [with T = CC1101]':
src/radio/RadioLibWrapper.h:137:3: warning: control reaches end of non-void function [-Wreturn-type]
   }
   ^
Compiling .pio/build/seeed_xiao_esp32c3/lib789/WiFi/WiFi.cpp.o
Compiling .pio/build/seeed_xiao_esp32c3/lib789/WiFi/WiFiAP.cpp.o
*** [.pio/build/seeed_xiao_esp32c3/src/main.cpp.o] Error 1
======================================================== [FAILED] Took 76.08 seconds ========================================================

Environment         Status    Duration
------------------  --------  ------------
seeed_xiao_esp32c3  FAILED    00:01:16.083
=================================================== 1 failed, 0 succeeded in 00:01:16.083 ===================================================
make: *** [Makefile:82: build] Error 1
```

phretor, 2023-04-28:
This is the offending line:

```
src/radio/../modules/../rfquack_logging.h:53:61: error: no matching function for call to 'HWCDC::begin(int, int)'
```

I'm assuming your boards is slightly different than a regular ESP32, so you may need to adjust the board definition, otherwise PlatformIO may load the wrong hardware abstraction layers.

woahotter, 2023-04-29:
tried `seeed_xiao_esp32c3` in the platformio.ini file, as that matches my board - gives `error: no matching function for call to 'HWCDC::begin(int, int)'`
tried `adafruit_qtpy_esp32c3` too just to test it, still gives me the same `error: no matching function for call to 'HWCDC::begin(int, int)'`

considering just buying an esp32 dev kit, but if you think i can do it with a esp32c3, please let me know

phretor, 2023-04-29:
You have to ensure that you can compile a "hello world" PlatformIO project for the esp32c3.

woahotter, 2023-04-29:
trying to do it on a windows machine, i was able to successfully flash a hello world project onto the esp32c3:
[image]

again, trying the rfquack build on same windows machine, it fails with the same HWCDC error - same on a linux machine

phretor, 2023-04-29:
I'm afraid there's some incompatibility with the HWCDC library and the esp32c3

Can you try to include it into your hello world project and see if the same error is triggered?

woahotter, 2023-04-29:
weird, works:
[image]

phretor, 2023-04-30:
Include would not suffice. To reproduce, you need to use the library in the same way it's used in (I think was) RadioLib, IIRC.

woahotter, 2023-05-03:
sorry for the late reply, doing my finals this week

i haven't found any decent documentation on HWCDC (what its used for, how to use it), can you point me in the right direction or can you give me an example?

phretor, 2023-05-03:
I never used it either, but you can copy from the code that triggers the error and include the minimum until you can reproduce the error in your standalone project.

whatotter, 2023-08-03:
okay, after fiddling with it for a bit now i just ripped out the code that gave me the issue and it worked:

in /src/rfquack_logging.h, i removed the if segment from `rfquack_logging_setup()` and it works now, and it flashed perfectly fine
i also disabled logging in the build.env

everytime i use the rfquack cli interface though, i get `'RFQuack' object has no attribute 'radioA'` after entering `q.radioA.help()`

whatotter, 2023-08-03:
actually, i may have found out why - how would i go upon changing the spi pins? the spi pins on the esp32c3 xiao are conflicting with gpio pin 9, which will make the esp go into download mode if its plugged in to the cc1101, which made me have to change it

it worked with another rf hacking tool just by changing the spi pins, but i cant seem to figure that out in here

phretor, 2023-08-04:
You can reassign pins as you would do in any PlatformIO project. Including SPI pins. If not advise to disable logging, tho.

whatotter, 2023-08-04:
> You can reassign pins as you would do in any PlatformIO project. Including SPI pins.
i haven't found it, and pio doesn't let you change spi pins - i meant as in one of the project files, like `rfquack.h`

> If not advise to disable logging, tho.
hopefully i can fix that, but i currently wanna get the base working and then fix