rfquack/RFQuack #50: Receiving not working on ESP32 DevKitv1 with cc1101

john19922, 2024-09-18:
Hi, I'm a newbie in ESP32 board. I tried to connect the board with cc1101, transmitting is working but receiving is not. Here is the pinout details, I tried to connect GDO0 with a different GPIO pin of ESP32 board but still the receiving is not working -

[image]

ESP32 board I'm using-
[image]

I'm not getting any receiving signal -
[image]

build.env configuration

RADIOA=CC1101
RADIOA_CS=5
RADIOA_IRQ=12

LOG_ENABLED=true

GUESSING_MODULE=true
FREQ_SCANNER_MODULE=true
MOUSE_JACK_MODULE=true
PACKET_FILTER_MODULE=true
PACKET_MOD_MODULE=true
PACKET_REPEAT_MODULE=true
ROLL_JAM_MODULE=true

platformio.ini config

[platformio]
default_envs = esp32devkit

Crsarmv7l, 2024-09-18:
I posted about that a long time ago, but basically you will need Syncwords in order to RX with this as it uses an older version of radiolib. 

In the more current versions of radiolib I submitted a few pulls one of which was to set PQT (Packet Quality Threshold) to 0 when using promiscuous mode. That enables an RX of everything flying around.

john19922, 2024-09-20:
I tried to edit 'int16_t CC1101::setPromiscuousMode(bool promiscuous) {'
based on the new Radiolib cc1101.cpp module but there is build error.

If you can please share more details of what modification do I need to do in Radiolib library it will really help or can share the working RX cc1101 Radiolib module.

Crsarmv7l, 2025-03-19:
I have already posted all the details needed both here and here: https://github.com/rfquack/RFQuack/issues/44#issuecomment-2096026675

I have written my own implementation due to the shortfalls. It won't be published until I fix a persistent hardware issue though.