rfquack/RFQuack #51: Troublesome installation and use of cli

maxdd, 2024-11-02:
Hello,
This is not really a bug but it's more related to the flow in using this project.
I'm currently struggling to find a somewhat flawless or the intended installation method and usage.
I have compiled and flashed my esp32devikit with make clean build flash and i get 

```
[RFQK]        134 T: Standalone mode network setup: done!
[RFQK]        235 T: Setting up serial transport
[RFQK]        246 T: Serial transport connected
[RFQK]        458 T: Frequency = 2400
[RFQK]        560 T: Registering modules...
[RFQK]        560 T: Module 'guessing' registered.
[RFQK]        561 T: Registering modules...
[RFQK]        561 T: Module 'frequency_scanner' registered.
[RFQK]        572 T: Registering modules...
[RFQK]        572 T: Module 'mouse_jack' registered.
[RFQK]        583 T: Registering modules...
[RFQK]        583 T: Module 'packet_filter' registered.
[RFQK]        584 T: Registering modules...
[RFQK]        595 T: Module 'packet_modification' registered.
[RFQK]        595 T: Registering modules...
[RFQK]        606 T: Module 'packet_repeater' registered.
[RFQK]        606 T: Registering modules...
[RFQK]        607 T: Module 'roll_jam' registered.
[RFQK]        617 T: Registering modules...
[RFQK]        618 T: Module 'ping' registered.
[RFQK]        618 T: Registering modules...
[RFQK]        629 T: Module 'radioA' registered.
[RFQK]        630 T: Setup is over.
[RFQK]        630 T: Transport is sending 2 bytes on topic RFQUACK/out/set/discovery/rfquack_CmdReply/discovery
<RFQUACK/out/set/discovery/rfquack_CmdReply/discovery~CAA=␀[RFQK]       1652 T: Starting main loop.
```
but the cli part is somewhat undocumented, the current repo says

`The CLI is now integrated in this repository. The old `rfquack/RFQuack-cli` repository is no longer used nor maintained.
`
but unless a setup.py isn't provided the rfq command will never work.
It also misses some dependencies as well (loguru).

So i had to fallback to https://github.com/rfquack/RFQuack-cli/tree/master and use it under docker but it seems like i should use the cli in client folder

maxdd, 2024-11-02:
The only reasonable solution i've found was

make docker-build-image
docker run --device /dev/ttyUSB0 --user=root --rm -it rfquack/rfquack bash
PYTHONPATH=$PYTHONPATH:$(pwd) python3 rfquack/cli.py

but i also get

```
NameError                                 Traceback (most recent call last)
File <ipython-input-6-e3c418729ffc>:5
      2 for freq in range (2405, 2474):
      3     q.radioA.set_modem_config(
      4     carrierFreq=freq)
----> 5     time.sleep(0.15)
      6 q.radioA.idle()

NameError: name 'time' is not defined
```

phretor, 2024-11-02:
I saw your comment on missing the `import time` ; did you solve this? If yes, do you mind sending a pull request?

maxdd, 2024-11-02:
If you are talking about the `import time` i generally type it right before starting interacting with the cli.