rfquack/RFQuack #61: ECRFv2 Instructions / Tips

jstockdale, 2025-02-05:
Ok I finally got rfquack running on the ecrf v2 and I can confirm it does work. Still testing but all three radios are seen and appear to function. I'd be happy to write it up if you want me to / if there's a good place to put it.

Obviously the first thing to do is build the firmware and flash it. Besides installing dependencies you also have to find replace "imp" in the j2 main cpp template with "importlib".

Then #1 if you get a error setting modulation it's probably because you installed rfquack from pip or the out of date RFQuack-cli repository. Instead set up a virtual env for python3 and then run  ```pipenv install -e .``` from inside the RFQuack repository root. The modulation issue will show up as OOK always failing to set. 

Issue #2: You also need to turn off one of the debug logging flags or you'll get errors running commands because it can't parse the spi messages with log statements. 

#3 there is a bug I'm still trying to figure out with the mouse jacking set radio assignment call. Hardcode the mouse jacking code to default to RadioC instead of RadioA. 

#4 if you find yourself having trouble with ipython you probably need to downgrade to 8.10.0 ... they broke the auto complete stuff for lots of packages after that.

If I think of anything else I'll add it here.

Thanks!! 😊

jstockdale, 2025-02-07:
Still can't get rx() working at all. tx() seems to work though. Seems like rx() not working might be a broader bug?

alanblake, 2025-03-19:
I'm trying to get RADIOC nRF24 rx() in  promiscuous mode using the ECRF build.env without success.

jstockdale, 2025-03-19:
I've had a lot of trouble getting rx() to work _at all_ on any of the
radios.

tx() seems to work fine on my devices.

I'm not sure how to fix but if I figure it out, I'll update this thread.

Cheers!

Crsarmv7l, 2025-03-19:
> Still can't get rx() working at all. tx() seems to work though. Seems like rx() not working might be a broader bug?

It's not. This uses FIFOs, FIFOs need a sync word match to be viewed as a valid signal.

Sync serial or async serial need to be used in order to rx anything and everything.